RobotLocomotion/drake #4606: mosek and gurobi don't work with xcode generator

RussTedrake, 2016-12-26:
both fail on the configure step in drake, with almost identical errors. 

```
/Users/russt/drake-distro/cmake/packages.cmake(54):  find_package(mosek ${dfp_required} ${dfp_quiet} ${DFP_UNPARSED_ARGUMENTS} )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(12):  set(PACKAGE_VERSION 7.1.0.45 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(14):  if(PACKAGE_VERSION VERSION_LESS PACKAGE_FIND_VERSION )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(16):  else()
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(18):  if(7.1.0.45 MATCHES ^([0-9]+)\\. )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(19):  set(CVF_VERSION_MAJOR ${CMAKE_MATCH_1} )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(24):  if(PACKAGE_FIND_VERSION_MAJOR STREQUAL CVF_VERSION_MAJOR )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(26):  else()
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(27):  set(PACKAGE_VERSION_COMPATIBLE FALSE )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(30):  if(PACKAGE_FIND_VERSION STREQUAL PACKAGE_VERSION )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(37):  if(${CMAKE_SIZEOF_VOID_P} STREQUAL  OR 8 STREQUAL  )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config-version.cmake(42):  if(NOT CMAKE_SIZEOF_VOID_P STREQUAL 8 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(1):  cmake_minimum_required(VERSION 3.1 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(8):  get_filename_component(PACKAGE_PREFIX_DIR ${CMAKE_CURRENT_LIST_DIR}/../../ ABSOLUTE )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(10):  macro(set_and_check _var _file )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(19):  include(CMakeFindDependencyMacro )
/usr/local/Cellar/cmake/3.7.0/share/cmake/Modules/CMakeFindDependencyMacro.cmake(20):  macro(find_dependency dep )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(21):  set(MOSEK_VERSION_MAJOR 7 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(22):  set(MOSEK_VERSION_MINOR 1 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(23):  set(MOSEK_VERSION_MINOR 0 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(24):  set(MOSEK_VERSION_TWEAK 45 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(25):  set(MOSEK_VERSION 7.1.0.45 )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(27):  find_dependency(Threads )
/usr/local/Cellar/cmake/3.7.0/share/cmake/Modules/CMakeFindDependencyMacro.cmake(21):  if(NOT Threads_FOUND )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(29):  set_and_check(MOSEK_INCLUDE_DIRS ${PACKAGE_PREFIX_DIR}/include/mosek )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(11):  set(MOSEK_INCLUDE_DIRS /Users/russt/drake-distro/xcode/install/include/mosek )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(12):  if(NOT EXISTS /Users/russt/drake-distro/xcode/install/include/mosek )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(30):  set_and_check(MOSEK_LIBRARY_DIRS ${PACKAGE_PREFIX_DIR}/lib )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(11):  set(MOSEK_LIBRARY_DIRS /Users/russt/drake-distro/xcode/install/lib )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(12):  if(NOT EXISTS /Users/russt/drake-distro/xcode/install/lib )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(31):  set(MOSEK_LIBRARIES mosek /usr/lib/libm.dylib Threads::Threads )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake(33):  include(${CMAKE_CURRENT_LIST_DIR}/mosek-c-targets.cmake )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-c-targets.cmake(6):  get_filename_component(PACKAGE_PREFIX_DIR ${CMAKE_CURRENT_LIST_DIR}/../../ ABSOLUTE )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-c-targets.cmake(10):  if(NOT TARGET Threads::Threads )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-c-targets.cmake(16):  if(NOT TARGET mosek )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-c-targets.cmake(17):  add_library(mosek SHARED IMPORTED )
/Users/russt/drake-distro/xcode/install/lib/mosek/mosek-c-targets.cmake(18):  set_target_properties(mosek PROPERTIES INTERFACE_LINK_LIBRARIES ${PACKAGE_PREFIX_DIR}/lib/libiomp5.dylib;/usr/lib/libm.dylib;Threads::Threads IMPORTED_LOCATION ${PACKAGE_PREFIX_DIR}/lib/libmosek64.7.1.dylib INTERFACE_INCLUDE_DIRECTORIES ${PACKAGE_PREFIX_DIR}/include/mosek )
/Users/russt/drake-distro/cmake/packages.cmake(58):  string(TOUPPER mosek dfp_package_upper )
/Users/russt/drake-distro/cmake/packages.cmake(60):  if(NOT mosek_FOUND AND FIND_PACKAGE_STRICT AND WITH_${dfp_package_upper} )
/Users/russt/drake-distro/cmake/packages.cmake(61):  message(FATAL_ERROR WITH_${dfp_package_upper} is enabled, but could NOT find mosek )
CMake Error at /Users/russt/drake-distro/cmake/packages.cmake:61 (message):
  WITH_MOSEK is enabled, but could NOT find mosek
Call Stack (most recent call first):
  /Users/russt/drake-distro/cmake/packages.cmake:148 (drake_find_package)
  CMakeLists.txt:15 (drake_find_packages)
```

jamiesnape, 2016-12-27:
They are working on the CI Xcode jobs: https://drake-jenkins.csail.mit.edu/view/Xcode/

Try
```
cd build
rm -rf gurobi-prefix mosek-prefix externals/gurobi externals/mosek
cmake .
xcodebuild
```

RussTedrake, 2016-12-27:
same error.

only difference I can see is that my `build` folder is named `xcode` here (since I also have a make build)

jamiesnape, 2016-12-27:
Try removing `xcode/drake-prefix` and `xcode/drake` also.

jamiesnape, 2016-12-27:
I just reproduced the error with a build folder called `xcode`. I will work on a fix.

RussTedrake, 2016-12-27:
it's strange that the cmake config files are clearly found (and parsed), but somehow WITH_* does not get set.

jamiesnape, 2016-12-27:
@RussTedrake can you check that `libmosek64.dylib` and `libgurobi60.so` are installed to `xcode/install/lib`?

RussTedrake, 2016-12-27:
yes.  both are present.

RussTedrake, 2016-12-27:
also just confirmed that it built ok when i made `build` the xcode build.

jamiesnape, 2016-12-27:
Can you send me the contents of the `xcode/install/lib/gurobi` and `xcode/install/lib/mosek` folders?

RussTedrake, 2016-12-27:
no gurobi folder in install lib.  

```
drake008% cat mosek/* > for_jamie.txt
drake008% pwd
/Users/russt/drake-distro/xcode/install/lib
```

```
drake008% ls
bot2-core							libdrakeRobotStateEncoder.dylib
ccd								libdrakeRobotStateLcmTypeUtil.dylib
cmake								libdrakeSchunkGripper.dylib
drake								libdrakeSchunkWsg.dylib
drakeConstructRigidBodyConstraint.mexmaci64			libdrakeSensors.dylib
drakeMexUtil.mexmaci64						libdrakeShapes.dylib
lcm								libdrakeSide.dylib
libBullet2FileLoader.2.83.dylib					libdrakeSplineGeneration.dylib
libBullet2FileLoader.dylib					libdrakeSpringMassSystemPlant.dylib
libBullet3Collision.2.83.dylib					libdrakeSystemAnalysis.dylib
libBullet3Collision.dylib					libdrakeSystemControllers.dylib
libBullet3Common.2.83.dylib					libdrakeSystemFramework.dylib
libBullet3Common.dylib						libdrakeSystemPrimitives.dylib
libBullet3Dynamics.2.83.dylib					libdrakeTrajectories.dylib
libBullet3Dynamics.dylib					libdrakeTrajectoryOptimization.dylib
libBullet3Geometry.2.83.dylib					libdrakeXMLUtil.dylib
libBullet3Geometry.dylib					libdrakeYAMLUtil.dylib
libBullet3OpenCL_clew.2.83.dylib				libdrakeZMPPlanner.dylib
libBullet3OpenCL_clew.dylib					libdrakeZMPTestUtil.dylib
libBulletCollision.2.83.dylib					libdrakeZMPUtil.dylib
libBulletCollision.dylib					libdrake_lcmtypes.dylib
libBulletDynamics.2.83.dylib					libf2c
libBulletDynamics.dylib						libf2c.a
libBulletInverseDynamics.2.83.dylib				libfcl.0.6.0.dylib
libBulletInverseDynamics.dylib					libfcl.0.6.dylib
libBulletSoftBody.2.83.dylib					libfcl.dylib
libBulletSoftBody.dylib						libgflags.2.2.0.dylib
libLinearMath.2.83.dylib					libgflags.2.2.dylib
libLinearMath.dylib						libgflags.dylib
libPythonQt.dylib						libgmock.dylib
libQtPropertyBrowser.dylib					libgmock_main.dylib
libbot2-core.1.dylib						libgtest.dylib
libbot2-core.dylib						libgtest_main.dylib
libbot2-lcmgl-client.1.dylib					libgurobi60.so
libbot2-lcmgl-client.dylib					libgurobi_c++.a
libbot2-lcmgl-renderer.1.dylib					libgurobi_g++4.2.a
libbot2-lcmgl-renderer.dylib					libiomp5.dylib
libbot2-param-client.1.dylib					liblcm-static.a
libbot2-param-client.dylib					liblcm.1.3.1.dylib
libccd.2.0.dylib						liblcm.1.3.95.dylib
libccd.2.dylib							liblcm.1.dylib
libccd.dylib							liblcm.a
libctkPythonConsole.1.0.dylib					liblcm.dylib
libctkPythonConsole.dylib					liblcmtypes_bot2-core.a
libddApp.dylib							liblcmtypes_bot2-core.dylib
libdrakeActuatorEffortToRigidBodyPlantInputConverter.dylib	liblcmtypes_bot2-lcmgl.a
libdrakeAtlasUtil.dylib						liblcmtypes_bot2-param.a
libdrakeAutomotive.dylib					liblcmtypes_bot2-procman.a
libdrakeAutomotiveLcm.dylib					libmosek64.7.1.dylib
libdrakeBouncingBall.dylib					libmosek64.dylib
libdrakeCollision.dylib						libmosekglb64.7.1.dylib
libdrakeCommon.dylib						libmosekjava7_1.jnilib
libdrakeControlUtil.dylib					libmosekscopt7_1.dylib
libdrakeControlledSpringMassSystemPlant.dylib			libmosekxx7_1.dylib
libdrakeConvexHull.dylib					libnlopt.0.9.0.dylib
libdrakeDynamicConstraint.dylib					libnlopt.0.dylib
libdrakeEstimators.dylib					libnlopt.dylib
libdrakeGeometryUtil.dylib					liboctomap.1.8.0.dylib
libdrakeHsrb.dylib						liboctomap.1.8.dylib
libdrakeIK.dylib						liboctomap.a
libdrakeIKoptions.dylib						liboctomap.dylib
libdrakeJoints.dylib						liboctomath.1.8.0.dylib
libdrakeKukaIiwaArm.dylib					liboctomath.1.8.dylib
libdrakeKukaIiwaArmCommon.dylib					liboctomath.a
libdrakeKukaIiwaWorld.dylib					liboctomath.dylib
libdrakeLCMSystem.dylib						librobotlocomotion-lcmtypes.dylib
libdrakeLCMSystem2.dylib					libsnopt_c.a
libdrakeLCMUtil.dylib						libspruce.dylib
libdrakeLcm.dylib						libtinyxml2.dylib
libdrakeLinearControl.dylib					libvtkDRCFilters.dylib
libdrakeMath.dylib						libvtkDRCFiltersPythonD.dylib
libdrakeMultibodyParsers.dylib					libyaml-cpp.0.5.3.dylib
libdrakeOptimization.dylib					libyaml-cpp.0.5.dylib
libdrakePendulumPlant.dylib					libyaml-cpp.dylib
libdrakeQPLocomotionPlan.dylib					mosek
libdrakeRBM.dylib						pkgconfig
libdrakeRBSystem.dylib						python2.7
libdrakeRigidBodyConstraint.dylib				rbmMexFunctions.mexmaci64
libdrakeRigidBodyPlant.dylib					robotlocomotion-lcmtypes
libdrakeRobotCommandToDesiredEffortConverter.dylib		snopt
libdrakeRobotStateDecoder.dylib
```

[for_jamie.txt](https://github.com/RobotLocomotion/drake/files/674497/for_jamie.txt)

jamiesnape, 2016-12-27:
Everything looks good for mosek from those files. Can you add the `NO_QUIET` argument to `drake_find_package()` here

https://github.com/RobotLocomotion/drake/blob/master/cmake/packages.cmake#L148

and send me any output relating to mosek?

RussTedrake, 2016-12-27:
gurobi get's hit first:
```
drake008% cmake .
CMake Warning at /Users/russt/drake-distro/cmake/packages.cmake:54 (find_package):
  Found package configuration file:

    /Users/russt/drake-distro/xcode/install/lib/cmake/gurobi/gurobi-config.cmake

  but it set gurobi_FOUND to FALSE so package "gurobi" is considered to be
  NOT FOUND.
Call Stack (most recent call first):
  /Users/russt/drake-distro/cmake/packages.cmake:144 (drake_find_package)
  CMakeLists.txt:15 (drake_find_packages)
```

RussTedrake, 2016-12-27:
same for mosek (after moving it ahead of gurobi in that list)
```
drake008% cmake .
CMake Warning at /Users/russt/drake-distro/cmake/packages.cmake:54 (find_package):
  Found package configuration file:

    /Users/russt/drake-distro/xcode/install/lib/mosek/mosek-config.cmake

  but it set mosek_FOUND to FALSE so package "mosek" is considered to be NOT
  FOUND.
Call Stack (most recent call first):
  /Users/russt/drake-distro/cmake/packages.cmake:144 (drake_find_package)
  CMakeLists.txt:15 (drake_find_packages)
```

RussTedrake, 2016-12-27:
also confirmed that I get to the end of gurobi-config.cmake (a status message prints fine there).

jamiesnape, 2017-01-05:
Can you try the setup I am using at the moment?
```
mkdir cmake-build-xcode
cd cmake-build-xcode
cmake -DWITH_GUROBI=ON -DWITH_MOSEK=ON -GXcode ..
xcodebuild
```

If that does not work, can you check out a fresh clone of drake and try the same there?

david-german-tri, 2017-04-11:
Closing as wontfix since no one is using xcode anymore TTBOMK.

closed 2017-04-11 by david-german-tri