RobotLocomotion/drake #4610: Spruce Reports Incorrect Working Directory in Bazel-based Unit Test

liangfok, 2016-12-27:
# Problem Definition

The following code is used to obtain the current working directory using `spruce`:

```c++
    spruce::path path(".");
    path.setAsCurrent();
    std::cout << "Current working directory: " << path.getStr() << std::endl;
```

The resulting print statement shows the current working directory returned by `spruce`:

```
Current working directory: /home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/bazel-sandbox/233c290e-6cc5-4b6d-ad9b-448462309355-0/execroot/drake-distro-1/bazel-out/local-opt/bin/drake/multibody/parsers/parser_common_test.runfiles/drake
```

The above-mentioned directory does not exist:

```
$ ls /home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/bazel-sandbox/233c290e-6cc5-4b6d-ad9b-448462309355-0/execroot/drake-distro-1/bazel-out/local-opt/bin/drake/multibody/parsers/parser_common_test.runfiles/drake
ls: cannot access '/home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/bazel-sandbox/233c290e-6cc5-4b6d-ad9b-448462309355-0/execroot/drake-distro-1/bazel-out/local-opt/bin/drake/multibody/parsers/parser_common_test.runfiles/drake': No such file or directory
```

jwnimmer-tri, 2016-12-27:
I suspect that at the time the test was running, such a directory _did_ exist.

Is the issue here just a request to get the new unit test working under Bazel?

liangfok, 2016-12-27:
Yes, I was trying to get `parser_common_test`  to run in #4576. Is there a way to pause or abort a Bazel build in such a way where all directories created by Bazel are kept intact?

liangfok, 2016-12-27:
Here is my WIP feature branch with instrumentation that replicates the above-mentioned error:

https://github.com/liangfok/drake/commit/179722dad661844a84987effcb479343417927b7

liangfok, 2016-12-27:
I received and applied @jwnimmer-tri's patch in #4576. Since the patch worked, I shall close this issue.

closed 2016-12-27 by liangfok