RobotLocomotion/drake #4611: snopt.hh const correctness and snopt BUILD file

jefesaurus, 2016-12-27:
It seems `snopt_solver.cc` uses `snopt.hh`, which at least in my unmodified copy of 7.2-8 has non-const pointers for the first arg of `snseti` and `snsetr`, which requires me to change your `snopt_solver.cc` to const cast. I'd rather make as few modifications to drake and snopt as possible. How did you get around this? Modifying the snopt signatures or some other loophole?

Also, I noticed you folks were adding bazel support, which I'm quite happy about. Turns out it's really simple to use Bazel to build SNOPT if you link to a system installed f2c. I've made a BUILD file for snopt 7.2-8 if you want inspiration. After adding the`const_cast`s in `snopt_solver.cc` it seems to just work with your tests.

```
package(default_visibility = ["//visibility:public"])

licenses(["notice"])

exports_files([
    "LICENSE",
])

# This links to system install f2c libraries:
# sudo apt-get install libf2c2-dev

filegroup(
    name = "snopt_c_sources",
    srcs = glob(["csrc/*.c"]),
)

cc_library(
    name = "snfilewrapper",
    hdrs = ["cppsrc/snfilewrapper.hh"],
    srcs = ["cppsrc/snfilewrapper.c"],
    linkstatic = 1,
    linkopts = ["-lf2c"],
    includes = ["cppsrc/"],
)

cc_library(
    name = "snopt",
    hdrs = ["cppsrc/snopt.hh"],
    srcs = [":snopt_c_sources"],
    deps = [":snfilewrapper"],
    includes = ["cppsrc/"],
    linkopts = ["-lf2c"],
    linkstatic = 1,
    copts = ["-Wno-parentheses", "-Wno-unused-but-set-variable"],
)

cc_binary(
    name = "csntoya",
    srcs = ["cexamples/toyfunction.h", "cexamples/toyfunction.c", "cexamples/snopt.h", "cexamples/snfilewrapper.h", "cexamples/csntoya.c"],
    deps = [":snopt"],
    includes = ["cexamples/"],
    linkstatic = 1,
    copts = ["-Wno-parentheses", "-Wno-unused-but-set-variable", "-Wno-unused-variable"],
)
```

If you do this, you could theoretically just have a `snopt.BUILD` in mainline drake, and then have the private repo in `WORKSPACES` reference the local build file. Would potentially make it a bit easier for third parties to sideload their own copies of snopt.

RussTedrake, 2016-12-27:
awesome.  assigning to the most relevant folks.  thanks!

jamiesnape, 2016-12-27:
Can you update the documentation to add `libf2c2-dev` and [install_prereqs.sh](https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/16.04/install_prereqs.sh) script? This will need changes to the CI images.

jamiesnape, 2016-12-27:
Also, we maintain the build system for SNOPT, so this could go into the SNOPT repo itself.

hongkai-dai, 2016-12-27:
@jefesaurus in https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/snopt_solver.cc#L368, we created a struct `SNOPTRun` to call `snseti` and `snsetr`. That struct is not stored inside `SnoptSolver`, so `SnoptSolver` is still const.

jefesaurus, 2016-12-30:
@hongkai-dai The specific issue I'm having is the `SNOPTRun` to snopt interface at 
https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/snopt_solver.cc#L151
https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/snopt_solver.cc#L160

```
  snopt::integer snSeti(std::string const& opt, snopt::integer val) {
    snopt::integer opt_len = static_cast<snopt::integer>(opt.length());
    snopt::integer err = 0;
    snopt::snseti_(opt.c_str(), &val, &iPrint, &iSumm, &err, D.cw.data(),
                   &D.lencw, D.iw.data(), &D.leniw, D.rw.data(), &D.lenrw,
                   opt_len, 8 * D.lencw);
    return err;
  }
```

The `SNOPTRun` function takes a const reference to the string `opt` and then passes it to the snopt function with `c_str()`, but at least in my version of snopt, the `snopt::snseti_(char *buffer, ...` function has no const arguments, which produces a compiler error for me when I build snopt_solver using bazel. So I can const cast, or use `-fpermissive`. Does your private snopt repo have a modified or newer snopt c interface with a `const char *buffer` in `snopt::snsetr_` and `snopt::snseti_` or is there something else I'm missing?

Also, there's one more wrinkle with using the system's libf2c: Under some circumstances I don't fully understand the shared object library complains with the following: 
```
glalonde@devenv:~/drake$ bazel run drake/solvers:mathematical_program_test 
INFO: Found 1 target...
ERROR: /code/drake/drake/solvers/BUILD:203:1: Linking of rule '//drake/solvers:mathematical_program_test' failed: linux-sandbox failed: error executing command /code/.cache/bazel/_bazel_glalonde/90ff29851e675113ecdf5b811c8cb48e/execroot/drake/_bin/linux-sandbox ... (remaining 14 argument(s) skipped).
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libf2c.so: undefined reference to `MAIN__'
collect2: error: ld returned 1 exit status
Target //drake/solvers:mathematical_program_test failed to build
```

Solution that appears to work from [SO](http://stackoverflow.com/questions/8345725/linker-errors-with-fortran-to-c-library-usr-lib-libf2c-so-undefined-referenc): Just force it to link against the static lib with something like this
```
cc_library(
    name = "snopt",
    hdrs = ["cppsrc/snopt.hh"],
    srcs = [":snopt_c_sources"],
    deps = [":snfilewrapper"],
    includes = ["cppsrc/"],
    linkopts = ["-l:libf2c.a"],
    linkstatic = 1,
    copts = ["-Wno-parentheses", "-Wno-unused-but-set-variable"],
)
```

hongkai-dai, 2016-12-30:
I see, Drake is using SNOPT 7.2-12, and the snseti_ takes a const char*. The signature in our version of SNOPT is 
```
void snset_
     ( const char *buffer, integer *iprint, integer *isumm,
       integer *inform__, char *cw, integer *lencw,
       integer *iw, integer *leniw,
       doublereal *rw, integer *lenrw,
       ftnlen buffer_len, ftnlen cw_len);
```

jwnimmer-tri, 2018-01-31:
Note that in #7240, I plan to rework a bit how we locate and build snopt.  Once that's merged, I suspect we could tweak it with version-sensing in `snopt_solver.cc` to admit different upstream signatures.

jwnimmer-tri, 2018-07-06:
I suspect that since we support SNOPT 7.6 now, that this issue is no longer relevant.  Please re-open with updated details in case not.

closed 2018-07-06 by jwnimmer-tri