RobotLocomotion/drake #4612: revisit LinearSystemSolver api

RussTedrake, 2016-12-27:
Thinking more about it, I think we need to improve the consistency of the LinearSystemSolver with the rest of MathematicalProgram.  I think that if one were to add (only) a bunch of linear constraints and call solve, then it should return infeasible if that system cannot be solved to tolerance.  In order to use the api as it’s currently implemented (which returns the minimum norm solution to Ax = b), we should force the user to instead add quadratic constraints of the form |Ax - b|^2.  

In my working branch, I’ve added a method to MathematicalProgram called AddL2NormCost(A,b) which supports that API.

edrumwri, 2016-12-27:
Since you're revisiting the API, do you want to consider a user-provided matrix structure description? Like saying that the matrix is symmetric, positive definite or triangular? The structure wouldn't even have to be guaranteed, but could provide a good starting point for a solution algorithm.

RussTedrake, 2016-12-27:
also a good idea.

hongkai-dai, 2017-08-08:
This is fixed. Now we report "kInfeasibleConstraints", if the linear system is infeasible.

closed 2017-08-08 by hongkai-dai