RobotLocomotion/drake #4623: Add operator overloading for symbolic::Variable

soonho-tri, 2016-12-27:
This is for https://github.com/RobotLocomotion/drake/issues/4619. We need operator overloading `+`, `-`, `*`, `/` between `symbolic::Variable`s. For example,

```c++
symbolic::Expression operator+(symbolic::Variable& lhs, symbolic::Variable& rhs)
```

soonho-tri, 2016-12-27:
/cc @hongkai-dai

jwnimmer-tri, 2016-12-27:
The cppguide says not do that, except when _Implicit conversions can sometimes be necessary and appropriate for types that are designed to transparently wrap other types. In that case, contact your project leads to request a waiver of this rule._

So in this case its borderline.  Likely a conversion from Variable to Expression is always desirable when demanded by the C++ expression.  But it's not quite a simple "transparent" type wrapper; it's a slightly different idea.

The downside of leaving it explicit is that there's a lot of operator-math boilerplate?  If we write that once and call it good, can we put it behind us, or will it always keep nagging for more?

soonho-tri, 2016-12-28:
@jwnimmer-tri , thanks for the comment. I'll write operator overloading. I think, thanks to today's discussion, we won't have different kinds of variables, and as a result, we don't need to worry about too-many overloadings..

soonho-tri, 2016-12-28:
@hongkai-dai , I'm working on it now. 

If this is blocking your workflow, please remove the `explicit` keyword at [drake/common/symbolic_expression.h#L143-L144]( https://github.com/RobotLocomotion/drake/blob/master/drake/common/symbolic_expression.h#L143-L144). It should work as if this issue is resolved.

closed 2016-12-28 by soonho-tri