RobotLocomotion/drake #4625: Styleguide should permit copy construction w/o assignment when assignment is impossible

sherm1, 2016-12-28:
Per [discussion](https://reviewable.io/reviews/robotlocomotion/drake/4616#-K_16R5tgir23Hdvk27K) in PR #4616: Our style guide Clarifications section requires use of const data members when possible, and the [Google style guide](https://google.github.io/styleguide/cppguide.html#Use_of_const) also encourages that discipline. Classes with const data members are suitable for copy & move construction, but cannot support copy & move assignment. However the [Google style guide](https://google.github.io/styleguide/cppguide.html#Copyable_Movable_Types) says:
> If you define a copy or move constructor, define the corresponding assignment operator, and vice-versa.

A literal reading would say you can't have const data members if you want to make a class copy constructible. I can discern no logical reason for that and I think the most likely interpretation is that Google forgot to add "if possible" to that requirement. I propose that we add that to our Clarifications section.

This is a pressing issue for us because literal adherence to "use const members" and "must be copy/move assignable if copy/move constructible" means that we currently can't even write a compliant POD class that can be put in an `std::vector<T>` since types `T` must be at least `MoveConstructible` . They don't need to be assignable at all.

Are there any objections to this clarification?

jwnimmer-tri, 2017-01-03:
I am skeptical of losing the "both" (in tandem) rule for copying.  I would like to see an example (or three) where this is needed before changing the guide.  I went back and the "define both in tandem" for copying dates back through at least 2008 in the cppguide, with various affirmative phrasings over time.  I don't think it's as simple as an oversight.

One reason I am hesitant is that it's possible that defining copy-ctor but deleting copy-assign has strange interactions with how locals, temporaries, copy elision, etc. operate.  Unfortunately I haven't yet found any write-up of this on the interwebs.  I can consult Meyers next time I have the books handy.

sherm1, 2017-01-03:
Wait, didn't you just do move-ctor-but-not-move-assignment in #4626? (Code in master is [here](https://github.com/RobotLocomotion/drake/blob/master/drake/systems/framework/system_port_descriptor.h#L62).) That is equally styleguide-nonconformant, but I think it was fine.

The compiler will be very upset if it depends on a method that didn't get generated. Are you worried that there is some undetected circumstance where bad things will happen?

jwnimmer-tri, 2017-01-03:
> Wait, didn't you just do move-ctor-but-not-move-assignment in #4626?

Yes.  I think Copy and Move could arguably be treated differently.  They take effect in different kinds of code, and I think Copy's much more broad applicability holds it to a higher standard.

I also think #4626 isn't the final answer either -- I think the port descriptor should either be fully copyable (without `const`s), or else fully noncopyable and or stored otherwise than an incrementally-assembled vector of values.  #4626 was a quick intermediate result that IMO is better than a halfway-copyable type.

> Are you worried that there is some undetected circumstance where bad things will happen?

I am concerned that code that looks like one thing ends up doing another, because what looks to a programmer like an assignment (because an `=` appears in the statement) actually turns into a copy.  Or that code that is working (due to using copies), no longer compiles after a minor perturbation for some new feature or refactoring such that assignment is required, whereas some copy-from-temporary used to be working.  And then we'd have to go and figure out how to make the type copy-assignable anyway, after we've already baked in a non-assignable assumption.  Better to just decide up-front and do it right, one way of the other.

---

But really, my main objection is procedural -- that we shouldn't override the guide until we have code that motivates the discussion.  And for copy, I haven't seen such code, nor can I imagine code where having a copy-only type is ever the best answer.

sherm1, 2017-01-04:
> But really, my main objection is procedural -- that we shouldn't override the guide until we have code that motivates the discussion.

Well, we kind of created this problem by requiring `const` for never-modified members. Assignment is always modification; construction is not.

I don't understand your distinction between copy and move -- they are treated identically in the style guide as best I can tell. Currently I think you are trying to solve an imaginary problem. Is there anything actually wrong with a class that can be copied or moved but not assigned to? Seems very reasonable to me.

jwnimmer-tri, 2017-01-04:
> Well, we kind of created this problem by requiring const for never-modified members. Assignment is always modification; ...

Careful: "requiring" could be misleading in that statement.  A `const` is only required when the member isn't changed after construction.  With an assignment operator present, the members can be changed after construction, so `const` is not required, even when the only post-construction change it ever sees is through assignment.

Thus, one internally consistent proposal is that: 
- copy-constructable types should have non-const members (and should also copy-assignable); 
- non-copyable types should have const members iff unchanged post-construction.

> ... is there anything actually wrong with a class that can be copied or moved but not assigned to?

Rather than spend time trying work out some counter-examples, I will again return the burden of proof to you.  When a cppguide edit is proposed, I think the burden should be on the proposer to justify the need for the change.

So again, is there a specific example or three you have at hand, where you want to have a copyable-but-not-assignable type?

sherm1, 2017-01-04:
> is there a specific example or three you have at hand, where you want to have a copyable-but-not-assignable type?

The example that immediately comes to mind is the one you wrote which is movable-but-not-move-assignable because it has const data members. I reject the idea that that is somehow distinct from copyable-but-not-copy-assignable for purposes of deciding what to do about the style guide, since they are treated identically there. But, since you ask [here](https://simbody.github.io/simbody-3.5-doxygen/api/classSimTK_1_1ArrayViewConst__.html#details) is a random example from a utility class in Simbody. It provides a const view into someone else's data; copy construction provides another view into that same data but copy assignment makes no sense so is suppressed. Isn't that what it ought to do?

I continue to believe that the style guide simply overlooked the possibility that assignment isn't always possible or reasonable. I think Drake reviewers should insist that a class with a copy/move constructor address copy/move assignment and provide assignment unless there is a compelling argument not to do so. I like having const members for data that shouldn't be changed after construction; that is a good reason not to have an assignment operator IMO. I don't really get why you are worried about this; it comes up now and again for perfectly good reasons. Assignment is much more demanding than construction because it requires editing an existing structure rather than creating a new one from whole cloth. Trying to create one when it isn't needed feels like jumping through unnecessary hoops.

jwnimmer-tri, 2017-01-04:
On first read, I find the ArrayViewConst link unconvincing on the general point.  Its fields do not appear to be const, so default-copy-assignment would work, and I see no reason why assignment is prima facie silly.  The class seems to have pointer semantics, so why not allow rebinding what the "pointer" refers to?

In any case, like everything in cppguide, the either-both-or-none is a guideline that can be broken and commented in the rare cases where its wrong.  Perhaps ArrayViewConst is one of them.  I remain unconvinced that we should add a specifically-documented local exception to encourage people to break the rule.

I suspect we will continue going in circles until there is Drake code on the table that cares.  Why should we add this kind of clarification now, rather than when we need it?

sherm1, 2017-01-04:
> Why should we add this kind of clarification now, rather than when we need it?

I only filed this issue because first *I* violated the style guide in PR #4616 (with copy), then *you* broke the style guide in PR #4626 (with move), each modifying the Input/OutputPortDescriptors. If you are happy leaving it that way with no style guide clarification, so am I. But the cat is already out of the bag.

jwnimmer-tri, 2017-01-04:
As I said upthread, I don't think #4286 should be the final answer.  It was an improvement, but still defective, as you say.

I would be content with an issue "PortDescriptor being move-constructable and not move-assignable is silently inconsistent with cppguide".  If it were up to me, I'd resolve that by fixing the code, not changing the guide.

sherm1, 2017-01-04:
I like what you did in #4626 and don't think it is defective. But I think we've beat the current issue to death and I'm happy to leave the style guide as is! Plus it is keeping you up too late :). Goodnight to you, and to this issue.

closed 2017-01-04 by sherm1