RobotLocomotion/drake #4635: lcm.bzl needs to parse .lcm files to discover the generated output file name

RussTedrake, 2016-12-29:
the current implementation assumes that the lcm filename corresponds to the structure name.  it need not (and does not in bot_core_lcmtypes).

see discussion at:
https://github.com/mwoehlke-kitware/bot_core_lcmtypes/pull/1

and https://github.com/RobotLocomotion/drake/pull/4608

jwnimmer-tri, 2016-12-29:
I believe the fast path to getting a build up and running would be to do like https://bazel.build/versions/master/docs/skylark/cookbook.html#custom-outputs and allow the `BUILD` file that invokes the `lcm_cc_library` macro to supply the `outs=[...]` list directly, overriding the default computation from `lcm.bzl`'s [_lcm_srcs_to_outs](https://github.com/RobotLocomotion/drake/blob/b8afe52b4aa5eb542a5c4581d7513b73929f5863/tools/lcm.bzl#L9).

If we do that, it would be nice if `outs=[]` was still defaulted to the obvious answer when we have a single LCM struct per file with matching name, so that `drake/lcmtypes/BUILD` could stay undisturbed.  (People who want artisinally-crafted source filenames that differ from the generated code's struct names can suffer, but the rest of us shouldn't have to.)

For the `bot_core_lcmtypes` case, there appears to be a mapping from filename to struct name, so the `outs=[]` be computed programmatically by the `robotlomotion_lcmtypes.BUILD` rule, so that populating the `outs=[]` list to pass into the revised `lcm_cc_library` macro shouldn't be very difficult.

---

Of course the most convenient solution is what's requested in the top post -- having the build system derive the output names by reimplementing its own LCM syntax parser, in order to learn what filenames to expect.  I believe that bazelbuild/bazel#900 (including its linked discussion) with respect to https://bazel.build/versions/master/docs/skylark/concepts.html#evaluation-model is relevant, in that doing that in Bazel is under-documented.

I don't immediately know how I would approach this.  I know with a repository rule I'd have a`repo_ctx`, at hand, so that when bringing an entire source tree into the workspace I could asses what struct definitions it contains.  (This is similar to how pulling in a pkg-config library works -- you sniff it out a workspace-loading time.)  But with either a macro or regular rule, I don't think I have the means to parse the source code and reason about it.

mwoehlke-kitware, 2016-12-29:
> having the build system derive the output names by reimplementing its own LCM syntax parser

JFTR, this sounds scarier than it needs to be. The "parser" I implemented for CMake (i.e. what LCM ships) is fairly simple-minded, which seems to be okay 99% of the time. This should be sufficient:

- Read the whole file into a string.
- Remove C and C++ comments (can do with a pair of regular expressions).
- Remove content in `{}`s (again, regular expression).
- Split the remainder at `;`s.
- Scan the resulting statements for ones that start with `package`, `enum` or `struct`.
  - For each such statement, split on whitespace; the second token needs to be acted upon.

mwoehlke-kitware, 2016-12-29:
In particular, see [this](https://gist.github.com/mwoehlke-kitware/a0e58920a1c0bb148df8cf91c5839d63) sample Python parser. The actual parsing is only about 10 LOC using nothing more complicated than a few regular expressions.

jwnimmer-tri, 2017-01-04:
FYI, I have _"I believe the fast path to getting a build up and running would be to [use] the mapping from filename to struct name ..."_ on a WIP branch, because libbot needed it.  I will likely get it into a PR sometime this week.

jwnimmer-tri, 2017-01-05:
#4684 adds support for `lcm_structs=[]` overrides in our `BUILD` files, so you can do stuff like (for `tools/libbot2.BUILD`):
```python
_LCM_SRCS = glob(["bot2-core/lcmtypes/*.lcm"])
_LCM_STRUCTS = [
    pathname[len("bot2-core/lcmtypes/bot_core_"):-len(".lcm")]
    for pathname in _LCM_SRCS]
lcm_java_library(
    name = "bot2-core-lcmtypes-java",
    lcm_package = "bot_core",
    lcm_srcs = _LCM_SRCS,
    lcm_structs = _LCM_STRUCTS,
)```

mwoehlke-kitware, 2017-01-05:
Is it possible to modify `lcm_java_library` (and friends? I assume there are similar functions for C/C++/Python bindings?) to expect fully qualified struct names instead of assuming a single package for everything?

jwnimmer-tri, 2017-01-05:
Yes, I think that's possible.  But what's the use case for having different packages in the same library?  Bazel leaf libraries are supposed to be tiny, and then aggregated into larger named units as needed.  So far, I haven't seen a Drake external that would want to mix multiple packages like that.  Is there an example to look at?

mwoehlke-kitware, 2017-01-05:
It's more a matter of a) paranoia, and b) parsing the `.lcm`s gives you fully qualified names anyway. What I'm thinking is if one of us can make that change, I can probably also add a function that takes the `.lcm` names and returns the list of fully qualified names. Then we can use that instead of making assumptions about the file name → struct name mapping, and won't need to specify the package name at all.

jwnimmer-tri, 2017-01-05:
If we go the route of parsing `*.lcm`s to discover struct and package name, then we can drop `lcm_package` and `lcm_structs` from the macro.  I'm still sure that doing that parsing is wise, or even possible in Bazel.

In the meantime, I don't see a reason to change the current macro signatures, since I believe they work well for all of the Drake-related cases I've seen.

jwnimmer-tri, 2017-01-09:
My #4684 approach is merged; de-assigning myself again.

david-german-tri, 2017-04-11:
> In the meantime, I don't see a reason to change the current macro signatures, since I believe they work well for all of the Drake-related cases I've seen.

I agree, and since no one has shown up to disagree further in the past three months, I'm going to close this as wontfix.  (Of course, it can be reopened if it is or becomes a pain point.)

closed 2017-04-11 by david-german-tri