RobotLocomotion/drake #4641: Update Drake To Support Various Combinations of SDF Pose Specifications

liangfok, 2016-12-29:
# Problem Definition

In #4615, several combinations of joint outboard frame, link frame, inertia frame, and body frame values were identified as being inconsistently supported or interpreted between Drake vs. Gazebo. This issue tracks the resolution of these discrepencies.

# Action Items

- [ ] simple_double_pendulum_l_is_not_specified.sdf: Works in both Drake and Gazebo, but differently. In Drake, the unspecified L frame is assumed to equal the mobilizer's outboard frame. In Gazebo, the L frame appears to be made equal to the model's frame.
- [ ] simple_double_pendulum_l_equals_i.sdf: Works in Gazebo only. We should update Drake to support this scenario.
- [ ] simple_double_pendulum_l_between_b_and_i.sdf: Works in Gazebo only. We should update Drake to support this scenario.

liangfok, 2017-04-12:
This is blocked on #4337.

jwnimmer-tri, 2019-07-02:
@EricCousineau-TRI Maybe you want to take over / triage / close this issue, as related to current senatics-related ongoing work?

EricCousineau-TRI, 2019-07-09:
Done. While the above examples might no longer apply, parity between Gazebo and Drake should be explicitly checked. At present, we have some deviations on MBP, so I'll repurpose this issue for that.

PR #12505 (2020-01-03): sdformat: Update to libsdformat 9.0 (SDFormat 1.7)
Resolves #4641

This PR incorporates libsdformat 9.0, which supports the specification of
SDFormat 1.7.

Specifically, this incorporates pose frame semantics for SDFormat, which are
described in this proposal:
http://sdformat.org/tutorials?tut=pose_frame_semantics_proposal&cat=pose_semantics_docs
(source permalink: https://bit.ly/2ZSl1kH)

A brief summary of the changes from the proposal:
* Users may now specify poses relative to another frame using
`//pose/@relative_to`
  * Before, this was `//pose/@frame`, and was not described in the spec a [... cut]
closed 2020-01-03 by EricCousineau-TRI