RobotLocomotion/drake #4642: Profile and speed up MathematicalProgram

hongkai-dai, 2016-12-30:
@RussTedrake mentions in the pose estimation problem, it takes longer to setup the problem, than solving the problem.

RussTedrake, 2016-12-30:
yes, for large numbers of samples.  (e.g. 5000, but 1000 might be enough).  it's probably not surprising since the process also consumes 12 GB of memory (presumably all being allocated dynamically!) through that setup time.  I think the amount of copying/allocating we do could potentially be substantially reduced -- for example, to add constant element-wise bounds to a big decision matrix variable, I have to allocate a constant vector of the same size.

hongkai-dai, 2017-01-03:
There are many dynamic memory allocation / copying in current optimization. I list some of them here

1. In each `Constraint` object, such as `LinearConstraint` which imposes `lb<=A_*x<=ub`, the matrix `A_, lb, ub` stored inside the constraint are all dynamic size, and copied from the constructor.
2. In `Binding` class, it binds a constraint with a **list** of decision variables. The decision variables are all stored in the dynamic-sized `DecisionVariableMatrixX` objects.

There is a quick fix that can partially solves the problem, that in `Binding` class, instead of binding a constraint with a **list** of `DecisionVariableMatrixX` objects, we can instead concatenate these `DecisionVariableMatrixX` objects into a single `DecisionVariableVectorX`. In the `Constraint` class, when calling the `Eval` function, we always need to concatenate the inputs into a single Eigen::VectorXd, so we can just directly bind constraint with a `DcisionVariableVectorX`. This fix will get rid of the slow `std::list` container.

hongkai-dai, 2019-03-26:
We need to create a benchmark.

jwnimmer-tri, 2023-05-22:
Once there is a benchmark provided (i.e., a reproduction recipe), someone can reopen or create a new issue.  For now, this ticket isn't adding any value.

closed 2023-05-22 by jwnimmer-tri