RobotLocomotion/drake #4643: Add Suite of Parsers Frames Unit Test With All Non-Zero Pose Values

liangfok, 2016-12-30:
# Problem Definition

In #4615, @amcastro-tri added a suite of unit tests for evaluating various model frame configurations, which resulted in several bugs being identified (#4641). @sherm1 pointed out that numerous frames employed in that suite were defined using `<pose>0 0 0 0 0 0</pose>`, which may result in additional bugs being hidden.

# Action Item

Add another suite of similar unit tests where the frame poses are all non-zero, which will hopefully root out additional frame-related bugs.

amcastro-tri, 2016-12-30:
Great idea @liangfok. Maybe this or another issue, but I think it'd also be a good idea to test the effect of `<use_parent_model_frame>` in `<joint>/<axis>`. The logic inside the SDF parser for this entry is quite obscure and I think we should cover its usage.

amcastro-tri, 2017-02-01:
cc'ing @clalancette. This had to do with a consistent unambiguous definition of frames.

rpoyner-tri, 2022-05-20:
This issue is so old, the SDFormat parser (and spec for that matter) have been completely rewritten. Closing.

closed 2022-05-20 by rpoyner-tri