RobotLocomotion/drake #4646: Add interpolation capability to the Simulator

sherm1, 2016-12-30:
(Per dicussion in #3734) Given a recorded trajectory containing continuous state values and derivatives retained at each integration time step, it is possible to interpolate between steps with no loss of accuracy, provided that the interpolation method is suitable for the integrator that was originally used to produce the trajectory. Any interpolant of matching order is tolerable, however the best ones are intimately entangled with the particular integration formula and may require extra evaluations or recorded information generated during the original simulation. Also, although we are currently using fixed-order integrators there are also good variable-order ones we may want to use at some point. Those use different interpolation formulas for different trajectory steps.

The Simulator should provide an interpolation API soon, even if initially suboptimal, so that we can improve it over time. The necessary information for optimal interpolation is not available elsewhere. 

Tasks:
- [ ] Report the integration method and (maximum) order so that a caller can at least use the right order for an independent interpolation. 
- [ ] Store the integration order used to produce each of the trajectory Context values from its predecessor.
- [ ] Provide a method- and order-sensitive interpolation capability to be used with recorded trajectories.
- [ ] Use the same interpolation method for integrator performance improvement (that is, use interpolated results to provide read-only sampling, such as Publish() calls, allowing larger steps).

References: 
- [Shampine, 1985](http://dx.doi.org/10.1137/0722060) provides basic discussion of interpolation for RK methods.
- [Gladwell, 1987](http://dx.doi.org/10.1137/0908038) has interpolation for RK Merson, discusses qualitative issues such as preserving monotonicity, and provides empirical measurements.
- The "cited by" section for these early papers is a good literature survey.

edrumwri, 2017-01-01:
@sherm1, your citations above seem to point to very specific interpolation mechanisms for individual integrators. Is it feasible to use a generic interpolation capability with some helper functions on the integrator side? Asked another way, does the interpolation capability need to exist within the individual integrators?

sherm1, 2017-01-01:
@edrumwri, in general interpolation should be built into the integrator for best performance and we should allow for that possibility. However, generic interpolation is also possible and will produce results with the right error *order*, but not necessarily with the same accuracy as the step end points. So I think we should provide some handy static methods in the IntegratorBase class, then delegate the actual interpolation to the integrator which *may* be implemented simply using the base class utilities but *could* do better than that. In practice we could get a long way with linear, quadratic, and cubic Hermite interpolation for now for the RK1, 2, and 3 we have at the moment. Simbody has some code we could grab for the cubic Hermite.

sammy-tri, 2017-01-05:
The description above says "Given a recorded trajectory containing continuous state values and derivatives retained at each integration time step..."  Does this mean part of the work for this ticket is to create a means to log the continuous state and the derivatives?  (I can see some ways to do this for state without changes to the simulator, but not for derivatives (well, not easily/cheaply/possibly safely).

sherm1, 2017-01-05:
> Does this mean part of the work for this ticket is to create a means to log the continuous state and the derivatives?

@sammy-tri, given a System and a Context it is always possible to recalculate the derivatives by calling `system.CalcTimeDerivatives(context, &derivs)`. So a trajectory that records the contexts as they come out of the Simulator can make use of derivatives for interpolation, even if they weren't saved.

sammy-tri, 2017-01-05:
Right, but where do you get access to the full context of the system being simulated?

Possibly this would need a wrapper system around the System/Diagram being simulated to cache the simulation context somewhere...  (passthrough functions with extra code in the passthrough Publish to log the context?  I might be missing something obvious though.)

sherm1, 2017-01-05:
Yes, that was my first thought too -- make an "every time step" Publish() method for the Diagram as a whole and use it to collect copies of the Context it receives. Currently we don't have a diagram-level Publish but it would be easy enough to add.

david-german-tri, 2017-01-05:
Why not just log Contexts from the Simulator, which naturally has access to them, and is already responsible for defining the timestep?

sherm1, 2017-01-06:
> Why not just log Contexts from the Simulator? 

That would be nicer, I agree. If we could get away with just an `std::vector<Context>` it would be very easy to do. I'm not sure that's adequate as a trajectory though since it wouldn't contain interpolation information. Also I think people will also want to save regularly-sampled trajectories rather than the irregularly-spaced and potentially huge variable ones. So we probably ought to design a SimulatorTrajectory class to get a proper trajectory out of the Simulator. Maybe just start with a hollow one containing an `std::vector` and some TODOs?

david-german-tri, 2017-01-06:
> So we probably ought to design a SimulatorTrajectory class to get a proper trajectory out of the Simulator. Maybe just start with a hollow one containing an std::vector and some TODOs?

Sounds right to me.  In particular:

> Also I think people will also want to save regularly-sampled trajectories rather than the irregularly-spaced and potentially huge variable ones.

This shouldn't be a difficult feature to add as a setting on Simulator, since we already have logic to evaluate which of several upcoming stop points is in force.

sammy-tri, 2017-01-06:
> Also I think people will also want to save regularly-sampled trajectories rather than the irregularly-spaced and potentially huge variable ones

I think this is a serialization question  If you're implementing SimulatorTrajectory and value() does the right thing for a given value of t, then I don't see a benefit to resampling.  I think I only see value to resampling if you want to export fixed-step data to some other tool (and I suspect you'd lose the integration information in the process if you did that?)

Another thought is that if you are going to implement SimulatorTrajectory, that there might not be anything left to do in #3734 which isn't now covered here.

sherm1, 2017-01-06:
> I only see value to resampling if you want to export fixed-step data to some other tool

I've seen it for visualization purposes where you only need 60Hz or whatever for later display. A variable step integrator can sometimes produce a *lot* of steps between 60Hz frames. But for now if we can capture all the steps we can always resample later.

edrumwri, 2017-01-07:
Producing dense outputs (or, at least, starting down the path to doing
this) is on my OKRs for this quarter- I'm watching this discussion closely.
I agree, Sam, that there does seem to be some significant intersection with
#3734.

--------------------------------
Evan Drumwright
Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

sherm1, 2018-08-14:
@edrumwri should this be revised or closed now?

edrumwri, 2018-08-14:
The current status related to this issue is:

1) We now have an integrator API for dense output sampling.
2) We have some specialized dense output sampling methods (a default dense sampling mechanism exists for integrators that do not provide specializations).
3) Simulator does not currently leverage the dense output.

edrumwri, 2018-10-08:
This issue appears to be fully resolved. Please reopen if you believe the disposition to be otherwise.

closed 2018-10-08 by edrumwri