RobotLocomotion/drake #4687: Need a Suite of Tests to Quantify Depth Sensor Performance

liangfok, 2017-01-05:
# Problem Definition

A simulated depth sensor may not have infinite performance due to its reliance on external software packages like [OpenGL](https://www.opengl.org/), [Bullet](http://bulletphysics.org/wordpress/), [OGRE](http://www.ogre3d.org/), and [VTK](http://www.vtk.org/). Another reason limiting performance is finite computational resources and wall-clock time. Since performance is finite, it would be helpful to have a suite of performance tests for evaluating different simulated depth sensors. This issue tracks the development of this test suite.

# Performance Tests to Implement

- [ ] Sensing bandwidth and minimum depth delta that can be distinguished at at what range.
- [ ] Smoothness of the resulting depth image when detecting various types of surfaces like a plane, sphere, cuboid, etc.
- [ ] CPU and memory overhead.

liangfok, 2017-01-05:
@hongkai-dai and @sherm1, I heard you previously discussed this. What other performance benchmarks should we implement?

hongkai-dai, 2017-01-06:
@sherm1 and @SeanCurtis-TRI were discussing the performance of Bullet in general, about how accurate it is.

liangfok, 2017-04-12:
I'm going to close this issue for now since it's not blocking anyone. I can imagine this topic being so deep that it cannot be easily tracked as a github issue.

closed 2017-04-12 by liangfok