RobotLocomotion/drake #4688: Update tinyxml-2

SeanCurtis-TRI, 2017-01-05:
TinyXML has recently been updated so that `XMLElement` instances know what line they came from.  This would be *immensely* helpful in parsing to print better error messages (e.g., "Your XML is crap on line 37.")  Should we update tinyxml so we can benefit from this?

jwnimmer-tri, 2017-01-05:
If its an easy port, then it sounds great!  If it's real work, then #4337 may be a better investment.

SeanCurtis-TRI, 2017-01-05:
Does OSRF's SDF parser handle our URDF parsing needs as well?  @amcastro-tri had asked the question in #4688.

liangfok, 2017-01-05:
It looks like OSRF's SDF parser has a way to convert URDF to SDF:

https://bitbucket.org/osrf/sdformat/src/a3fa3d1163cc2fe991ac106e51695b99f8b119ca/include/sdf/parser_urdf.hh?at=default&fileviewer=file-view-default

SeanCurtis-TRI, 2017-01-05:
And that would make the `sdf::SDF` class our intermediate representation for both URDF and SDF?

jwnimmer-tri, 2017-01-05:
Maybe?  I had a TODO to discuss this with relevant folks in the next week or two.  I think it would be at least the first IR.  Whether we have another IR between it and the multibody runtime would be up for discussion.

SeanCurtis-TRI, 2017-01-05:
At first blush, I would imagine there could easily be another IR.  Essentially, everything that happens in `RBT::compile` would need to be represented in whatever IR we use.

liangfok, 2017-01-05:
It looks like the TinyXML2 was updated to support line numbers only 12 days ago: https://github.com/leethomason/tinyxml2/commit/c3a19156ffe05f793831d76dbe0183ab8d1e1846

Thus, `libsdformat`'s [built-in copy of TinyXML](https://bitbucket.org/osrf/sdformat/src/a3fa3d1163cc2fe991ac106e51695b99f8b119ca/src/win/tinyxml/?at=default) will need to be updated to get it.

SeanCurtis-TRI, 2017-01-05:
Which also implies that `libsdfformat` doesn't provide error messages with line numbers either.

liangfok, 2017-01-05:
Yes, that is true. I'm currently thinking updating which version of TinyXML2 Drake is using will be a relatively trivial change relative to replacing Drake's parsers with OSRF's parser.

sherm1, 2017-01-05:
+@nkoenig for questions above about libsdfformat ...

jwnimmer-tri, 2017-01-05:
> Thus, libsdformat's built-in copy of TinyXML will need to be updated to get it.

No, that's only for Windows.

> Or, we will need to ensure the system version of TinyXML has the update ...

No, Drake uses its own copy of tinyxml, not the system one.

liangfok, 2017-01-05:
I see, `libsdformat` uses the system version of TinyXML when built on Linux:

https://bitbucket.org/osrf/sdformat/src/a3fa3d1163cc2fe991ac106e51695b99f8b119ca/CMakeLists.txt?at=default&fileviewer=file-view-default#CMakeLists.txt-97

I wonder if that means it'll find the version Drake is using.

jwnimmer-tri, 2017-01-05:
This is about two derivatives away from the actual topic at hand.

liangfok, 2017-01-05:
+1 for updating Drake's version of TinyXML.

david-german-tri, 2017-04-11:
@SeanCurtis-TRI, I've removed this from Kitware's queue, and will leave it to you to prioritize.

SeanCurtis-TRI, 2017-04-11:
Do we have any sense of timeline for the OSRF sdf parser landing?

liangfok, 2017-04-11:
/cc: @clalancette ^^

jwnimmer-tri, 2018-12-21:
Per https://github.com/RobotLocomotion/drake/blob/df86de15c0ba63a616dd347b263d8e2cc083875d/tools/workspace/tinyxml/repository.bzl, Drake uses the OS-provided version of tinyxml2.  To avoid conflicts with other ecosystem software, we should strongly prefer to keep it this way.  If the problem becomes acute, we can look into compiling our own local copy with hidden visibility or new names.  In the meantime, we can close this until we have a need.

closed 2018-12-21 by jwnimmer-tri