RobotLocomotion/drake #4691: Unable to Add Simple Debug Statements Due to AutoDiffScalar Template Type

liangfok, 2017-01-05:
# Problem Definition

When a class is templated on `Eigen::AutoDiffScalar` among other types, we cannot instrument its methods with simple print statements. For example, try adding the following log statement to `kinematics_cache-inl.h`, in `KinematicCacheElement`'s constructor's body:

```
  drake::log()->debug("KinematicsCacheElement: transform_to_world: {}",
      transform_to_world);
```

You will experience the following compile-time error:

```
/home/liang/dev/drake-distro-2/build/install/include/spdlog/fmt/bundled/ostream.h:91:12: error: invalid operands to binary expression ('std::basic_ostream<char>' and 'const Eigen::Transform<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, 73, 1> >, 3, 1, 0>')
    output << value;
    ~~~~~~ ^  ~~~~~
```

This is unfortunate since having the ability to instrument the code with debug statements often facilitates the debugging process. Note that the above error persists even if we're trying to debug a use-case of `KinematicsCacheElement<double>` instead of `KinematicsCacheElement<Eigen::AutoDiffScalar>`.

david-german-tri, 2017-01-05:
One possible solution: we could provide a to-string function for supported scalar types, which selects an implementation using SFINAE.

jwnimmer-tri, 2017-01-05:
I don't even think we need that fancy, we just need to define one of the standard stringizer free functions for each of our scalar types.

amcastro-tri, 2017-01-05:
> we just need to define one of the standard stringizer free functions for each of our scalar types.

you mean `operator<<`?  that'd be awesome. However in this case I think the problem is that we do not have `operator<<` defined for `Isometry3`.

jwnimmer-tri, 2017-01-05:
Or an ADL version of `std::to_string`, which IIRC spdlog would also use?