RobotLocomotion/drake #4693: BUILD file in drake-distro root conflicts with "from source" instructions

RussTedrake, 2017-01-05:
Just went to make a fresh build from master on a different (mac) laptop, and hit this one.

```
31-34-10% cd drake-distro
31-34-10% mkdir build
mkdir: build: File exists
```

nice.  i can't actually create a build directory right now.  and given some assumptions in our cmake files (#4606), I can't name it something else.

SeanCurtis-TRI, 2017-01-05:
Hard to imagine where the build folder came from; it doesn't appear in master.

sherm1, 2017-01-05:
There is, OTOH, a *file* named `BUILD` in master. Is this possibly a platform where file names are case-insensitive so that `BUILD` and `build` name the same file?

RussTedrake, 2017-01-05:
I'm sorry.  I thought all of that was implied.  Yes -- on mac the filenames are case insensitive, so the existence of the BUILD file in drake-distro disallows the creation of a build directory.

jamiesnape, 2017-01-05:
Just hit that myself today. All our infrastructure uses a directory called `build` so it is somewhat of an issue. OS X CI builds delete `BUILD` (they assume it was a left over `build` directory).

jamiesnape, 2017-01-05:
This may also be what is caused the update column of CDash to turn red for OS X builds.

jwnimmer-tri, 2017-01-05:
Understood.  I should be able to get a work-around PR in tonight.  Local work-around would be to build using CMake (with rm BUILD; mkdir build) xor Bazel.

jwnimmer-tri, 2017-01-06:
Directly on point: bazelbuild/bazel#552.

closed 2017-01-06 by jwnimmer-tri

jamiesnape, 2017-01-06:
> Is there a reasonably cost-effective way we could have made CI find this for us (even post-merge / nightly)?

It is tricky because cleaning the `build` directory was itself a safety feature in case removal of the workspace from the previous build failed. CDash did pick up that something was wrong, but we do not currently consider failures computing the change log an error. I will give it some thought.

PR #4696 (2017-01-06): Rename top-level BUILD file to avoid confusing OSX
Fixes #4693.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4696)