RobotLocomotion/drake #4700: Consider Storing State Vector Index Information in DrakeJoint

liangfok, 2017-01-06:
# Problem Definition

Currently, state vector index information is stored in `RigidBody`. This complicates the process by which users obtain a joint's state within the state vector since it forces the user to either first obtain the joint's outboard `RigidBody`, or use a method like the currently-buggy `RigidBodyTree::computePositionNameToIndexMap()` (see #4697).

# Suggested Course of Action

Per discussion in #4698, we should consider moving and/or copying this information into `DrakeJoint`. This would allow Drake to provide the following API for obtaining state vector index semantics:

```
const DrakeJoint& RigidBodyTree::GetJointByName(const std::string& name, int model_instance_id);
int DrakeJoint::get_index(std::string& position_name) const;
```

Having the above API would allow a user to execute:

```
int index_in_state_vector = rigid_body_tree->GetJointByName("Joint4", 1 /* model_instance_id */).get_index("Joint4_x");
```

jwnimmer-tri, 2017-01-06:
RBT is dead.  Why do we care to document its deficiencies?

liangfok, 2017-01-06:
I suppose this goes back to the question raised here: https://github.com/RobotLocomotion/drake/issues/4633#issuecomment-270674395. It sounds like improvements made to RBT _could potentially_ find its way into the new MBT since RBT is being used as a source of inspiration for MBT.

sherm1, 2017-01-06:
My inclination would be to not fix this in RBT unless it is blocking. If not blocking I think it will be more productive to lobby @amcastro-tri to make sure it is considered early in MBTree.

liangfok, 2017-01-06:
Great! Since @amcastro-tri was the original source of inspiration for this issue (see [this comment](https://reviewable.io/reviews/robotlocomotion/drake/4698#-K_oPTpX0D-CRYeqv9A-)), I will assume that he will definitely take it into consideration early in MBTree's design.

closed 2017-01-06 by liangfok