RobotLocomotion/drake #4711: Need to Restrict Drake Build To Use Super-build Versions of Octomap and FCL

liangfok, 2017-01-09:
# Problem Definition

When building Drake using ROS Indigo's `catkin`, the build will fail if `ros-indigo-fcl` and `ros-indigo-octomap` are installed. Here is the error, which was first reported in #3814:

```
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/src/geometry/octree/octree.cpp:38:
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:179:16: error: no member named 'getNodeChild' in 'octomap::OcTree'
  return tree->getNodeChild(node, childIdx);
         ~~~~  ^
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:191:16: error: no member named 'getNodeChild' in 'octomap::OcTree'
  return tree->getNodeChild(node, childIdx);
         ~~~~  ^
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:203:16: error: no member named 'nodeChildExists' in 'octomap::OcTree'
  return tree->nodeChildExists(node, childIdx);
         ~~~~  ^
/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree-inl.h:214:16: error: no member named 'nodeHasChildren' in 'octomap::OcTree'
  return tree->nodeHasChildren(node);
         ~~~~  ^
```

In #4709, the installation instructions were updated to remove `ros-indigo-fcl` and `ros-indigo-octomap`, to force Drake to use its own versions of these two libraries. This is sub-optimal since many other ROS packages like MoveIt! depend on these two packages. To allow both to co-exist, we should modify Drake's build system to prioritize its versions of FCL and Octomap over ROS Indigo's versions.

# Possible Course Of Action

@jamiesnape mentions that the solution will be something like adding the following to Drake's `CMakeLists.txt#:224`:

```
-Dfcl_DIR:PATH=${CMAKE_INSTALL_PREFIX}/lib/cmake/fcl
```

liangfok, 2017-01-09:
Here are the locations of the FCL and Octomap `-config files:

```
$ cd drake_catkin_workspace
$ find . -name "fclConfig.cmake"
./build/drake/externals/fcl/fclConfig.cmake
./build/drake/externals/fcl/src/CMakeFiles/Export/lib/cmake/fcl/fclConfig.cmake
$ find . -name "octomap-config.cmake"
./src/drake/externals/octomap/octomap/lib/cmake/octomap/octomap-config.cmake
./install/share/octomap/octomap-config.cmake
./build/drake/externals/octomap/InstallFiles/octomap-config.cmake
```

I instrumented `drake_catkin_workspace/src/drake/CMakeLists.txt` with the following print statement:

```
message(STATUS "******** CMAKE_INSTALL_PREFIX = ${CMAKE_INSTALL_PREFIX}")
```

Here is the result:

```
******** CMAKE_INSTALL_PREFIX = /home/liang/dev/drake_catkin_workspace/install
```

I then tried to add the following line to `drake_catkin_workspace/src/drake/CMakeLists.txt`:

```
    -Doctomap_DIR:PATH=${CMAKE_INSTALL_PREFIX}/share/octomap
```

Here is the commit: https://github.com/liangfok/drake/commit/84a2a2b41128250f67ce1d0082adf90c50026e27

Unfortunately the problem persisted.

I next tried to include the argument along with `catkin config` as follows:

```
$ cd drake_catkin_workspace
$ catkin config --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DWITH_DREAL=FALSE -DDISABLE_MATLAB=TRUE -Doctomap_DIR:PATH=`pwd`/install/share/octomap
```

Unfortunately the problem persisted.

jamiesnape, 2017-01-09:
When you say the problem persisted after you made the changes, was that with a clean build?

liangfok, 2017-01-09:
Oops. No, I will try a clean build now. Thanks!

liangfok, 2017-01-09:
I did a [clean build](http://drake.mit.edu/from_source_ros.html#how-to-perform-a-clean-build) using the modified `CMakeLists.txt` file as described in https://github.com/liangfok/drake/commit/84a2a2b41128250f67ce1d0082adf90c50026e27 and the problem persisted. Here is the first error indicating that the ROS Indigo version of Octomap was being found:

```
cd /home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl/src && /usr/bin/clang++-3.9   -Dfcl_EXPORTS -I/home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl/include -I/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include -I/home/liang/dev/drake_catkin_workspace/install/include/eigen3 -isystem /home/liang/dev/drake_catkin_workspace/install/include  -g -fPIC   -std=c++11 -W -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-delete-non-virtual-dtor -Wno-overloaded-virtual -Wno-unknown-pragmas -Wno-deprecated-register -mfpmath=sse -msse -msse2 -msse3 -mssse3 -o CMakeFiles/fcl.dir/narrowphase/continuous_collision.cpp.o -c /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/src/narrowphase/continuous_collision.cpp
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/src/narrowphase/continuous_collision.cpp:38:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/continuous_collision-inl.h:47:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/collision.h:68:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/collision-inl.h:43:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix.h:86:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix-inl.h:73:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/detail/traversal/octree/collision/mesh_octree_collision_traversal_node.h:46:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree.h:48:
In file included from /opt/ros/indigo/include/octomap/octomap.h:37:
In file included from /opt/ros/indigo/include/octomap/OcTree.h:38:
In file included from /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.h:44:
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.h:262:11: warning: class member cannot be redeclared [-Wredeclared-class-member]
    class leaf_iterator;
          ^
/opt/ros/indigo/include/octomap/OcTreeIterator.hxx:260:11: note: previous declaration is here
    class leaf_iterator : public iterator_base {
          ^
```
I also tried using the following command line technique to no avail:

```
$ catkin config --cmake-args -DCMAKE_BUILD_TYPE:STRING=Debug -DWITH_DREAL=FALSE -DDISABLE_MATLAB=TRUE -Doctomap_DIR:PATH=`pwd`/install/share/octomap
```

liangfok, 2017-01-10:
Here is the result:

https://gist.githubusercontent.com/liangfok/5af7eb224683743e36613c84f8576150/raw/0530985896e2177dd84ca2b62707139c7e6b0307/drake_build_trace_debug_fcl_using_ros_octomap.txt

Line 18 shows the use of CMake flag `--trace`.

Line 71211 shows the addition of `fcl` as an external. It does **not** include the flag `-Doctomap_DIR:PATH=${{CMAKE_INSTALL_PREFIX}}/share/`. I believe this is why `fcl` uses ROS Indigo's version of octomap below.

Line 91229 shows the addition of `drake` as an external. It includes the use of `-Doctomap_DIR:PATH=${{CMAKE_INSTALL_PREFIX}}/share/`. This is the only place where `-Doctomap_DIR` shows up.

Line 130852 shows FCL incorrectly using ROS Indigo's version of Octomap.

liangfok, 2017-01-10:
I am now trying a [clean build](http://drake.mit.edu/from_source_ros.html#how-to-perform-a-clean-build) using the `--trace` flag and the following modification:

https://github.com/liangfok/drake/commit/b9be9caf2bab6c61b4b41850d0db46cb8cf7abef

Hopefully this'll do the trick!

liangfok, 2017-01-10:
Here is the result:

https://gist.github.com/liangfok/061dbef16fc93c00de68de7009be8693

Line 71210 shows the addition of `fcl` as an external using the correct `-Doctomap_DIR` flag:

```
/home/liang/dev/drake_catkin_workspace/src/drake/CMakeLists.txt(151):  drake_add_external(fcl PUBLIC CMAKE CMAKE_ARGS -DFCL_BUILD_TESTS=OFF -Doctomap_DIR:PATH=${{CMAKE_INSTALL_PREFIX}}/share/octomap DEPENDS ccd eigen octomap )
```

Line 130797 shows the command for running `cmake` on `fcl` using the correct `-Doctomap-DIR` flag:

```
cd /home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl && /home/liang/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY:=ON -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY:=ON "-DCMAKE_PREFIX_PATH:=/home/liang/dev/drake_catkin_workspace/install;/usr" -DCMAKE_INSTALL_PREFIX:PATH=/home/liang/dev/drake_catkin_workspace/install -DCMAKE_BUILD_TYPE:STRING=Debug -DBUILD_SHARED_LIBS:BOOL=ON -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang-3.9 -DCMAKE_C_FLAGS:STRING= -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++-3.9 -DCMAKE_CXX_FLAGS:STRING= -DCMAKE_EXE_LINKER_FLAGS:STRING= -DCMAKE_MODULE_LINKER_FLAGS:STRING= -DCMAKE_SHARED_LINKER_FLAGS:STRING= -DCMAKE_STATIC_LINKER_FLAGS:STRING= -DJava_JAVA_EXECUTABLE:FILEPATH=/usr/bin/java -DJava_JAVAC_EXECUTABLE:FILEPATH=/usr/bin/javac -DJava_JAVAH_EXECUTABLE:FILEPATH=/usr/bin/javah -DJava_VERSION_STRING:=1.7.0_121 -DLIB_SUFFIX:STRING= -DFCL_BUILD_TESTS=OFF -Doctomap_DIR:PATH=/home/liang/dev/drake_catkin_workspace/install/share/octomap "-GUnix Makefiles" /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl
Re-run cmake no build system arguments
-- The CXX compiler identification is Clang 3.9.1
-- The C compiler identification is Clang 3.9.1
-- Check for working CXX compiler: /usr/bin/clang++-3.9
-- Check for working CXX compiler: /usr/bin/clang++-3.9 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/clang-3.9
-- Check for working C compiler: /usr/bin/clang-3.9 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- FCL uses Octomap
-- Configuring done
-- Generating done
-- Build files have been written to: /home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl
```

Looking in `/home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl/CMakeCache.txt`, I see the following two lines indicating `fcl` should be using the correct version of `octomap`:

```
//Dependencies for the target
fcl_LIB_DEPENDS:STATIC=general;/home/liang/dev/drake_catkin_workspace/install/lib/liboctomap.so;general;/home/liang/dev/drake_catkin_workspace/install/lib/liboctomath.so;general;ccd;

//No help, variable specified on the command line.
octomap_DIR:PATH=/home/liang/dev/drake_catkin_workspace/install/share/octomap
```

Unfortunately, line 130854 shows that the wrong version of `octomap` is still being used:

```
cd /home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl/src && /usr/bin/clang++-3.9   -Dfcl_EXPORTS -I/home/liang/dev/drake_catkin_workspace/build/drake/externals/fcl/include -I/home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include -I/home/liang/dev/drake_catkin_workspace/install/include/eigen3 -isystem /home/liang/dev/drake_catkin_workspace/install/include  -g -fPIC   -std=c++11 -W -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wno-delete-non-virtual-dtor -Wno-overloaded-virtual -Wno-unknown-pragmas -Wno-deprecated-register -mfpmath=sse -msse -msse2 -msse3 -mssse3 -o CMakeFiles/fcl.dir/narrowphase/continuous_collision.cpp.o -c /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/src/narrowphase/continuous_collision.cpp
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/src/narrowphase/continuous_collision.cpp:38:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/continuous_collision-inl.h:47:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/collision.h:68:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/collision-inl.h:43:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix.h:86:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/detail/collision_func_matrix-inl.h:73:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/narrowphase/detail/traversal/octree/collision/mesh_octree_collision_traversal_node.h:46:
In file included from /home/liang/dev/drake_catkin_workspace/src/drake/externals/fcl/include/fcl/geometry/octree/octree.h:48:
In file included from /opt/ros/indigo/include/octomap/octomap.h:37:
In file included from /opt/ros/indigo/include/octomap/OcTree.h:38:
In file included from /opt/ros/indigo/include/octomap/OccupancyOcTreeBase.h:44:
/opt/ros/indigo/include/octomap/OcTreeBaseImpl.h:262:11: warning: class member cannot be redeclared [-Wredeclared-class-member]
    class leaf_iterator;
          ^
/opt/ros/indigo/include/octomap/OcTreeIterator.hxx:260:11: note: previous declaration is here
    class leaf_iterator : public iterator_base {
          ^
```

jamiesnape, 2017-01-11:
Catkin is doing bad things to `CMAKE_PREFIX_PATH` it seems. I am assuming if you source `/opt/ros/indigo/setup.bash`, but then do a regular build (using pure CMake, no Catkin) everything works out fine.

liangfok, 2017-01-11:
@jamiesnape: Thanks for the suggestion. I'm now trying what you suggest (source ROS, then build using CMake):

```
$ source /opt/ros/indigo/setup.bash
$ echo $ROS_ROOT
/opt/ros/indigo/share/ros
$ echo $CMAKE_PREFIX_PATH
/opt/ros/indigo
$ cd drake-distro; mkdir build; cd build
$ cmake .. -G Ninja -DCMAKE_BUILD_TYPE:STRING=Debug -DDISABLE_MATLAB=TRUE -DWITH_DREAL=FALSE -DWITH_SNOPT=FALSE --trace
$ ninja -v -j1
```
I'll update this post with the results!

jamiesnape, 2017-01-11:
Yes, so after some more digging, CMake is choosing the correct version of OctoMap and does not have `/opt/ros/indigo` in any of its configuration files. Why the compiler insists on using the ROS version, I am still investigating.

liangfok, 2017-01-11:
I believe we've finally root-caused the problem. It relates to this comment: https://github.com/RobotLocomotion/drake/issues/3814#issuecomment-271128929. Drake's version of FCL and Octomap are being included using `-isystem` instead of `-I` to get around the fact that at least one FCL header contains code that violates the `-Wshadow` error. When ROS is sourced, it sets environment variable `CPATH` equal to the following:

```
$ echo $CPATH
/opt/ros/indigo/include
```

Apparently, `CPATH` has higher priority than `-isystem` resulting in the compiler finding ROS' versions of FCL and Octomap instead of Drake's versions.

One way to get around the problem is to unset `CPATH`:

```
$ unset CPATH
```

This works because Drake proper doesn't actually use any of ROS' libraries. I suspect this will not work when there are other packages in the workspace that do rely on ROS libraries. Maybe one workaround is to first build package `drake` with `CPATH` unset.  Then re-build the workspace with `CPATH` set. I will continue to test.

liangfok, 2017-01-14:
I tried adding a `unset(ENV{CPATH})` to the top of Drake superbuild's `CMakeLists.txt` to no avail. As @jamiesnape mentioned, this probably did not work because the error is build-time, not configure-time.

jwnimmer-tri, 2017-03-05:
I think this is low priority, because it is catkin-only, doesn't affect our supported configurations, and is likely moot once BazelDrake is a superbuild project, and octomap & fcl are not.

PR #6783 (2017-08-15): Remove ROS from within Drake
Closes #3418, closes #3429, closes #4711, closes #5579, and closes #6674.

Relates in part to #5012.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/6783)
closed 2017-08-15 by sammy-tri