RobotLocomotion/drake #4719: Tracking issue for OS X-specific development cost

david-german-tri, 2017-01-09:
This issue is for gathering data on PRs that incur OS X-specific development cost. It is not a forum for discussing the implications. Examples of "cost" include:

Developer effort spent resolving OS X-specific issues.
Delayed merge due to discussing or resolving OS X-specific issues.
OS X-specific flakes in pre-merge continuous integration.
Rollbacks or forward fixes due to failures in OS X post-merge continuous integration.

Where possible, please reference an issue or PR.

david-german-tri, 2017-01-09:
#4717 has demonstrated that we need to reproduce in Bazel the platform-specific compiler flag selection logic that we've had in CMake for a while.  Specifically, we need to add flags with clang/OS X spellings and gcc/Linux spellings into respective CROSSTOOL stanzas, where if we only supported one platform we could just `.bazelrc` them.

Two hours so far today.

rpoyner-tri, 2017-01-09:
Spent maybe 2 hours working out OSX details for #4701.

david-german-tri, 2017-01-25:
Grappling with adding IPOPT to Bazel (#4896 with dynamic libraries, #4905 with static libraries).  Both approaches work with Linux, and both approaches have their own unique OS X gremlins.  Dynamic libraries trigger some kind of deeply confusing loader path problem, which might be a bazel bug and might just be my fault: https://github.com/bazelbuild/bazel/issues/507#issuecomment-275128269

Static libraries get us past that problem, but then we discover that `-lgfortran` is only available on Mac via the homebrew `gcc` package, which in turn requires a wacky `-L` path.  This path is somehow scribed into `pkg-config` files at ipopt build time.

6 hours so far.

liangfok, 2017-01-30:
I'm estimating #4956 incurred 1 hour of developer time mostly due to context switching overhead. 

To my knowledge, the Linux compilers chose the correct `abs()` and only the OSX compiler chose the wrong one.

david-german-tri, 2017-01-30:
In #4905, I added a new test rule that only works on Linux, leading to build breakage on OS X.   I've spent two hours so far this morning trying to figure out how to disable a test on OS X only.

david-german-tri, 2017-01-30:
#4970 - there went another day of my life, and it's still a kludge.  Thanks, Homebrew.

david-german-tri, 2017-03-02:
I spent at least 20 hours developing enough background knowledge to write #5334.

david-german-tri, 2017-03-02:
#5362 broke the build and wasted several people's time because we didn't 
(and maybe still don't) fully understand the subtleties of a Mach-O specific linker optimization relating to exception unwinds.  Perhaps 2 or 3 engineer hours so far.

david-german-tri, 2017-03-20:
About 5 hours understanding how `launchd` interacts with environment  variables, leading to #5570.

david-german-tri, 2017-03-21:
In #5155, we chose to link Bazel builds statically by default, in part because dynamic linking on OS X is different, and a little complicated, and I hadn't figured out how to make it work.  I didn't expect anyone to be disk-constrained in 2017, but it turns out some VM users are.  Not sure how to put an estimate on this.

closed 2017-04-03 by david-german-tri