RobotLocomotion/drake #4733: Check if constraint is violated at some solution value

hongkai-dai, 2017-01-10:
The API should be
```
bool Binding::CheckConstraintSatisfied(const Eigen::Ref<const Eigen::VectorXd>& x, double tol = 1E-14)
```

hongkai-dai, 2017-01-10:
@RussTedrake

RussTedrake, 2017-01-11:
My original request was for use in unit tests.  Might this also pave the way for one of the most important features that we're still missing in mathematical program -- the solver reporting out which constraints are infeasible?  We could obviously do it for snopt (and presumably all of the nonlinear solvers which carry around a "best" solution).  Are the APIs sufficient so we could also do it for gurobi/mosek?

I'm knee-deep in debugging an infeasible mccormick envelope from a random test case, and it's hard to find which constraint is to blame!

hongkai-dai, 2017-01-11:
In Gurobi, it can return the set of self-contradicting constraints, they explain the API here 
https://nathanbrixius.wordpress.com/2012/09/28/detecting-the-sources-of-model-infeasibility-using-gurobi/
http://www.gurobi.com/documentation/6.5/examples/diagnose_and_cope_with_inf.html

Mosek has some procedure to automatically repair an infeasible problem, by relaxing each constraints, and penalizing the amount of relaxation. The approach is described here http://docs.mosek.com/7.0/toolbox/Primal_feasibility_repair.html

I can add API to support returning these information.

RussTedrake, 2017-01-11:
that would be amazing.  thanks hongkai!  (i know your list keeps getting longer!)

EricCousineau-TRI, 2017-04-14:
## Quick Overview

Check if constraint is satisfied / violated (and possibly expose the degree of violation).

Example Usage:

    auto constraint = make_shared<BoundingBoxConstraint>(...);
    return constraint->CheckSatisfied(x);

## Schedule

Next. Should precede #3130.

## Functionality

* Provide ability to check if constraint is satisfied / violated
    * Violation value
        * `< 0` - Satisfied
        * `== 0` - Satisfied, active (if inequality)
        * `> 0` - Violated
        * Equality constraints will always return a violation `>= 0`
* Save computation by leveraging existing solver result information / results when available

## Implementation Plan

* New Methods:
    * Default (expensive) implementation, for unittesting:
        * `Constraint::`
            * `protected: Vector EvalViolation(Vector x)`
                * *Alternative*: Make `public`
            * `public: virtual bool CheckSatisfied(Vector x, tol)`
                * Default Implementation: `return (EvalViolation(x) < tol).all()`
            * Could pave way to directly checking if it is active, e.g. `(EvalViolation(x).abs() < tol).all()`

    * To be completed for #3130:
        * Rationale for waiting for next issue:
            * If a Solver fails, the current implementation will not return the intermediate solution (typical pattern is `if (!good) return k$FAILURE; else SetDecisionVariableValues(...)`
                * Examples: `IpoptSolver_NLP::finalize_solution`, `GurobiSolver::Solve`, `MosekSolver::Solve`
                * Counterexample: `SnoptSolver::Solve`
                * Will need to either (a) store intermediate / failed solution or (b) directly query infeasible constraints
        * `MathematicalProgram::`
            * `public: bool CheckBindingSatisfiedAtSolution(Binding<C> binding, double tol = default_eps)`
                * Best: When available, use dual variables and complementarity from specific solver
                * Default: Use direct computation

## Precedents:

* `drake/matlab/solvers/NonlinearProgram.m`
    * Functionality: `[infeasible_constraint_name] = infeasibleConstraint(obj,x)`
        * Will evaluate and compare to see if constraints are violated by checking for positive constraint values

closed 2017-05-04 by EricCousineau-TRI