RobotLocomotion/drake #4745: Need Documentation On When It's Safe To Call LeafSystem::DeclareFoo().

liangfok, 2017-01-11:
The documentation for the following `LeafSystem` methods should be updated to indicate when it's safe to call them:

* DeclareUpdatePeriodSec()
* DeclarePeriodicUpdate()
* DeclarePublishPeriodSec()
* DeclareContinuousState()
* DeclareDiscreteState()

For example, I think there may be a requirement that these methods only be called prior to the start of the simulation. If this is true, this fact should be documented.

liangfok, 2017-01-11:
Assuming the above methods should not be called after the simulation starts, can this be enforced via exception throwing?

liangfok, 2017-01-11:
Oh, I just realized that `LeafSystem` is a System 2.0 construct and shall thus reassign this issue.

david-german-tri, 2017-01-11:
These methods aren't `const`.  They should be safe to call whenever you have a mutable handle on a `LeafSystem`.  If that's not true, it's a bug.

> Assuming the above methods should not be called after the simulation starts, can this be enforced via exception throwing?

No, absolutely not!  `System`s are models.  They compute outputs as a function of contexts.  They do not store application state, and indeed are usable for applications other than simulation.  The phrase "after the simulation starts" is meaningless to a System.

closed 2017-01-11 by david-german-tri