RobotLocomotion/drake #4746: Can LeafSystem Provide Corresponding Accessors to the Various DeclareFoo() Methods?

liangfok, 2017-01-11:
Leaf System provides the following methods for use by sub-classes:

* DeclareUpdatePeriodSec()
* DeclarePeriodicUpdate()
* DeclarePublishPeriodSec()
* DeclareContinuousState()
* DeclareDiscreteState()

I've just encountered a use-case where I'd like to get the current state that's being set by these methods. For example:

* GetUpdatePeriodSec()
* GetPeriodicUpdate()
* GetPublishPeriodSec()
* GetContinuousState()
* GetDiscreteState()

Can these methods be added?

sherm1, 2017-01-11:
Seems reasonable. I'd use the "cheap" syntax, like `get_update_period_sec()` for accessors.

david-german-tri, 2017-01-11:
@liangfok It's possible, but can you be more specific about the problem you're actually trying to solve?

> GetUpdatePeriodSec()
> GetPeriodicUpdate()
> GetPublishPeriodSec()

There might be more than one, so the return structure would have to describe that.

> GetContinuousState()
> GetDiscreteState()

Again, the return structure would have to be fairly descriptive: q/v/z dimensions, vector type, etc.

liangfok, 2017-01-11:
> Can you be more specific about the problem you're actually trying to solve?

When unit testing `LcmPublisherSystem`'s ability to , I'd like to verify that the desired next update time is correct. See #4731, lcm_publisher_system_publish_rate_test.cc.

liangfok, 2017-01-11:
I decided to remove the aforementioned check from #4731 and shall thus close this issue.

closed 2017-01-11 by liangfok