RobotLocomotion/drake #4747: Consider Making 0 Instead of +Inf Signify "Update Every Sim Time Step"

liangfok, 2017-01-11:
Currently, `std::numeric_limits<T>::infinity()` is used to signify to the simulator that a system wants to be updated every major time step of the simulation. This is awkward since it could also be interpreted to mean "never update me" or "update me at time infinity". Consider using a value of 0 to signify the need to update a system every major time step of the simulation. It would be less confusing since, assuming simulation time >= 0, that would mean, "update me now" or "update me yesterday already".

For background context, see @jwnimmer-tri's comment here: https://reviewable.io/reviews/RobotLocomotion/drake/4731#-KaAtj8mLcrcdtXypmiU

jwnimmer-tri, 2017-01-11:
@liangfok Which method in particular?  I seem to recall that the infinity-semantics in System2 were okay, and only the PR draft was confusing.

liangfok, 2017-01-11:
Sorry, the method I'm referring to is `System::DoCalcNextUpdateTime()`:

https://github.com/RobotLocomotion/drake/blob/0210c74af7a5b8129388ab92e04c856fefcfe61f/drake/systems/framework/system.h#L851

jwnimmer-tri, 2017-01-11:
Yeah, that one is no problem IMO.  When the question is "When should I be next updated?" the answer "Never" is perfect.

liangfok, 2017-01-11:
But does "Never" imply "Update at the next major time step"?

jwnimmer-tri, 2017-01-11:
Nope.  Updates only happen when you ask.

Note that "Publish every major timestep" is an entirely different concept.

liangfok, 2017-01-11:
Oh I see. OK. I suppose I'm still getting mixed up between the use of `System<T>::DoCalcNextUpdateTime()` vs. `LeafSystem<T>::DeclarePublishPeriodSec()` to throttle the publishing of LCM messages.

closed 2017-01-11 by liangfok