RobotLocomotion/drake #4748: Errors provisioning LLVM on Xenial (Ubuntu 16.04)

SeanCurtis-TRI, 2017-01-11:
For a sequence of nights, starting 1/5/17, two nightly builds failed to run successfully:

   - [linux-xenial-unprovisioned-gcc-nightly-release](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-xenial-unprovisioned-gcc-nightly-release/)
   - [linux-xenial-unprovisioned-clang-nightly-release](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-xenial-unprovisioned-clang-nightly-release/)

The problem appears to be the fact that the LLVM apt repo is not currently accessible. 

As of 1/10/17, there appeared to be no notice on the LLVM email lists, and this was confirmed as an issue independent of the CI by attempting to `apt update` on a Xenial computer.

jamiesnape, 2017-01-11:
Possibly
https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/16.04/install_prereqs.sh#L25
https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/16.04/install_prereqs.sh#L31
need to be updated to
```
deb http://apt.llvm.org/xenial/  llvm-toolchain-xenial-3.9 main
```
See http://apt.llvm.org.

jamiesnape, 2017-01-11:
Also https://github.com/RobotLocomotion/drake/blob/master/drake/doc/ubuntu_trusty.rst#c-compiler would need updating.

jwnimmer-tri, 2017-01-11:
Yes, the new URL works.  Patch forthcoming.

jwnimmer-tri, 2017-01-11:
Spoke too soon, not actually working yet.

jwnimmer-tri, 2017-01-11:
Yeah, http://apt-origin.llvm.org/xenial/dists/llvm-toolchain-xenial/InRelease exists but http://apt-origin.llvm.org/xenial/dists/llvm-toolchain-xenial-3.9/InRelease is missing.  I guess LLVM is just broken?

I'm punting on this.

wjwwood, 2017-01-20:
Should this be closed now?

jwnimmer-tri, 2017-01-20:
Probably.  Tagging Betsy to be sure.

BetsyMcPhail, 2017-01-20:
Yes, thank you, fixed via #4800.

closed 2017-01-20 by BetsyMcPhail