RobotLocomotion/drake #4749: CI spuriously declares build failure in Bazel builds

jwnimmer-tri, 2017-01-11:
The Bazel CI builds appear to be using CTest's default error regexps and fail the build when one of those matches, even if Bazel returned success (0) exitcode.

If I'm understanding correctly, it looks like `CTEST_CUSTOM_ERROR_MATCH`'s default value at https://cmake.org/gitweb?p=cmake.git;a=blob;f=Source/CTest/cmCTestBuildHandler.cxx;h=3bf54f0f455f23b341dc4806c74fce32c491b8a6;hb=HEAD#l26 is breaking my CI build.  How do we make `CTEST_CUSTOM_ERROR_MATCH` regexps either disabled, or never fail the build?  (The exitcode should be the only thing we look at in CI, not whether a number surrounded by colons ever appeared in the console output.)  I tried `set(CTEST_CUSTOM_ERROR_MATCH "")` in `drake-distro/drake/CTestCustom.cmake.in` but that didn't seem to help.

Perhaps in the Bazel CI we don't have `CTEST_USE_LAUNCHERS = ON`, which is what nerfs the error regexps in the other builds.  https://itk.org/Bug/print_bug_page.php?bug_id=10099

/CC @david-german-tri 

I will post a link to a sample PR failure soon.

jwnimmer-tri, 2017-01-11:
I expect that #4750 will demonstrate the failure once it finishes building.

jwnimmer-tri, 2017-01-11:
For reference, I first discovered this in #4737 and worked around it with ce129e3.

jamiesnape, 2017-01-12:
Closed via RobotLocomotion/drake-ci#71.

closed 2017-01-12 by jamiesnape