RobotLocomotion/drake #4752: Bazel build error using Ubuntu Trusty

liangfok, 2017-01-11:
I know Ubuntu Trusty is not (may never be) officially supported, but I thought I'd document this in the hope someone will know a fix or workaround.

# The Error

```
ERROR: /home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/external/glib/BUILD:5:1: Linking of rule '@glib//:lib' failed: linux-sandbox failed: error executing command /home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/execroot/drake-distro-1/_bin/linux-sandbox ... (remaining 19 argument(s) skipped).
clang: error: unsupported option '-pass-exit-codes'
```

jwnimmer-tri, 2017-01-11:
Why is `clang` appearing there in the output?  Was your `CXX` set to clang for some reason?

liangfok, 2017-01-11:
Yes, I had the following in my `~/.bashrc`:

```
export CC=clang-3.9 CXX=clang++-3.9 FC=gfortran-4.9 F77=gfortran-4.9
```

I will change it to the following and try again:

```
export CC=gcc-4.9 CXX=g++-4.9 FC=gfortran-4.9 F77=gfortran-4.9
```

liangfok, 2017-01-11:
After switching to `gcc`, I get the following error:

```
$ bazel build //...
INFO: Found 638 targets...
ERROR: /home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/external/gtest/BUILD:3:1: C++ compilation of rule '@gtest//:main' failed: linux-sandbox failed: error executing command /home/liang/.cache/bazel/_bazel_liang/ede03c0a430a52111efe35db021d2956/execroot/drake-distro-1/_bin/linux-sandbox ... (remaining 53 argument(s) skipped).
ccache: failed to create /nonexistent/.ccache (No such file or directory)
INFO: Elapsed time: 2.959s, Critical Path: 0.48s
```

david-german-tri, 2017-01-11:
Regarding the gcc error, is `ccache` on your path?   http://drake.mit.edu/bazel.html

Regarding the clang error, it's very possible that affects Xenial as well.  I'm working in the area today anyway, so I'll take a look.

liangfok, 2017-01-11:
Thanks. Yes, `ccache` was in my path. Will try again with it removed. Thanks!

liangfok, 2017-01-11:
The build completed. Thanks again for the help!

closed 2017-01-11 by liangfok