RobotLocomotion/drake #4755: Add Expand method for symbolic expression

soonho-tri, 2017-01-11:
Example : `pow((x + y), 2).Expand()` -> `pow(x, 2) + 2 * x * y + pow(y, 2)`

/cc @hongkai-dai

jwnimmer-tri, 2017-01-11:
This is just to produce something in polynomial form (like https://github.com/RobotLocomotion/drake/blob/master/drake/common/polynomial.h#L16), or are there other kinds of Expansions beyond a spotless polynomial that we want to Expand to?

Because if only the former, having a "ToPolynomial()" method that either returns a Polynomial or fails-fast might be better?  (I'm reading between the lines here a bit.)

hongkai-dai, 2017-01-11:
@jwnimmer-tri , `ToPolynomial()` will be great!

soonho-tri, 2017-01-11:
OK, I'll add the following function in `symbolic::Expression` class:

```c++
namespace symbolic {
class Expression {
...
Polynomial ToPolynomial() const;  // throws if it's not a polynomial expression.
...
};
}  // namespace symbolic

jwnimmer-tri, 2017-01-11:
Sounds nice.  Though I wonder if calling code can be sure it's supposed to pass?  If not, we might need `bool IsPolynomial` and/or have it return `optional<Polynomial>`.

soonho-tri, 2017-01-11:
@jwnimmer-tri , good point! I'll add the following one as well:

```c++
bool IsPolynomial() const;
```

soonho-tri, 2017-01-16:
@hongkai-dai , FYI : A variable in `Polynomial` is `unsigned int` while a variable in `Expression` has both of `id_` (`size_t`) and `name_` (`std::string`). I'll make sure that after translation from an `Expression` to a `Polynomial`, a variable in `Polynomial` still keeps the same `ID`. However, the name part will be gone.

Do you see any problem caused by this loss of information in your application?

soonho-tri, 2017-01-16:
FYI2, `name_ : std::string` in a `symbolic::Variable` is treated as extra information in that `operator==`, `operator<`, and `get_hash` only use `id_` and ignore `name_`.

jwnimmer-tri, 2017-01-16:
I was imagining that [`VariableNameToId`](https://github.com/RobotLocomotion/drake/blob/master/drake/common/polynomial.cc#L534) and [`IdToVariableName`](https://github.com/RobotLocomotion/drake/blob/master/drake/common/polynomial.cc#L558) would still be relevant.

For example, that only `symbolic::Variable` with a name in the magic spotless msspoly varname set (<= 4 characters from supported set) can be converted to `Polynomial` form.  Or perhaps that with an opt-out the variable names get randomized but the bijection map between `VarType` and `symbolic::Variable::name` is also returned the caller (or possibly supplied by the caller).

hongkai-dai, 2017-01-16:
The name is typically used for debugging purpose. So if we do not have a name in polynomial, I am OK with it now. We can add this debugging capability back later.

I think @jwnimmer-tri 's suggestion to use `VariableNameToId` and `IdToVariableName` is a good solution. But if this takes a while to implement polynomial variable with a string name, then I think we can put it into a feature PR.

soonho-tri, 2017-01-20:
Closed by #4821.

closed 2017-01-20 by soonho-tri