RobotLocomotion/drake #4767: Drake build failed with `source directory already configured` in ipopt configure

rdeits, 2017-01-12:
I wasn't able to build drake master on Ubuntu 14.04 because configuration of ipopt failed with:

```
[81/207] Performing configure step for 'ipopt'
FAILED: ipopt-prefix/src/ipopt-stamp/ipopt-configure 
cd /home/rdeits/locomotion/drake-distro/build/externals/ipopt && /home/rdeits/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -E env CC=/usr/local/bin/cc CFLAGS= CXX=/usr/local/bin/c++ CXXFLAGS= LDFLAGS= F77=/usr/bin/gfortran FC=/usr/bin/gfortran FFLAGS= CC=/usr/local/bin/cc CXX=/usr/local/bin/c++ /home/rdeits/locomotion/drake-distro/externals/ipopt/configure --prefix=/home/rdeits/locomotion/drake-distro/build/install --disable-shared --includedir=/home/rdeits/locomotion/drake-distro/build/install/include/ipopt --with-blas=BUILD --with-lapack=BUILD --with-pic && /home/rdeits/tools/cmake-3.5.2-Linux-x86_64/bin/cmake -E touch /home/rdeits/locomotion/drake-distro/build/ipopt-prefix/src/ipopt-stamp/ipopt-configure
checking build system type... x86_64-unknown-linux-gnu
checking whether we want to compile in debug mode... no
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether /usr/local/bin/cc accepts -g... yes
checking for /usr/local/bin/cc option to accept ANSI C... none needed
configure: C compiler options are: -O3 -pipe -DNDEBUG -Wimplicit -Wparentheses -Wsequence-point -Wreturn-type -Wcast-qual -Wall -Wno-unknown-pragmas -Wno-long-long  
checking whether we are using the GNU C++ compiler... yes
checking whether /usr/local/bin/c++ accepts -g... yes
checking whether C++ compiler /usr/local/bin/c++ works... yes
configure: C++ compiler options are: -O3 -pipe -DNDEBUG -Wparentheses -Wreturn-type -Wcast-qual -Wall -Wpointer-arith -Wwrite-strings -Wconversion -Wno-unknown-pragmas -Wno-long-long  
configure: Trying to determine Fortran compiler name
checking whether we are using the GNU Fortran 77 compiler... yes
checking whether /usr/bin/gfortran accepts -g... yes
configure: Fortran compiler options are: -O3 -pipe  
checking for egrep... grep -E
checking whether ln -s works... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
configure: error: source directory already configured; run "make distclean" there first
ninja: build stopped: subcommand failed.
```

(this happened with the unix makefiles generator as well). Nuking the entire `build` folder was not sufficient to solve the problem. 

Instead, I had to delete the entire `externals/ipopt` folder and re-checkout that path. I'm posting here so that the fix is searchable in the future.

liangfok, 2017-01-12:
Any idea what state your `drake-distro` directory was in prior to this error occurring? For example, did it initially build OK but only start to fail after, for example, rebasing with `master`?

rdeits, 2017-01-12:
I hadn't run a build in several weeks, and had just pulled the latest master.

jamiesnape, 2017-01-12:
Probably before ipopt became an out-of-source build. Running `make distclean`as the error suggested should be sufficient.

jamiesnape, 2017-01-12:
Note that `ipopt` always uses `make` regardless of the generator as it is an Autotools project.

kunimatsu-tri, 2017-01-12:
I ran into the same problem and the workaround deleting externals/ipopt didn't work for me.
I've also tried building from scratch and this didn't work neither.

```
$ git clone https://github.com/RobotLocomotion/drake.git
$ cd drake
$ mkdir build
$ cd build
$ cmake ..
$ make -j24
```

Any suggestion?

jamiesnape, 2017-01-12:
Not sure how a failure due to an already configured source directory would be possible on a build from a freshly cloned repo, especially as master does not actually configure the source directory anymore. What is the output of this?
```
cd drake/externals/ipopt
git status
```

liangfok, 2017-01-12:
Just curious whether using `ninja` suffers the same problem:

```
$ git clone https://github.com/RobotLocomotion/drake.git
$ cd drake
$ mkdir build
$ cd build
$ cmake ..  -G Ninja
$ ninja
```

kunimatsu-tri, 2017-01-12:
> What is the output of this?

git status output at drake/externals/ipopt is:
```
HEAD detached at 11649b7
nothing to commit, working directory clean
```

jamiesnape, 2017-01-12:
It the source directory has not been configured then. If it had, you would see:
```
$ git status
HEAD detached at 11649b7
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	Ipopt/Makefile
	Ipopt/config.log
	Ipopt/config.status
        ....
```
Are you sure that was the error?

kunimatsu-tri, 2017-01-12:
Sorry, I have misunderstood. 
Building from scratch worked for me.
When I tried it from scratch, the problem was #4711, not this issue.

When I commented out source /opt/ros/indigo/setup.bash, it worked.

closed 2017-01-12 by rdeits

liangfok, 2017-01-12:
Should this issue be added to the FAQ? http://drake.mit.edu/faq.html

jamiesnape, 2017-01-12:
Maybe, though I would not expect it to affect many more users given that `ipopt` began to be built out-of-source just over two months ago.

liangfok, 2017-01-13:
I just encountered this on OSX and worked around it by doing a comprehensive nuke (#4551). Deleting `externals/ipopt` did not work for me. I neglected to try `make distclean`.