RobotLocomotion/drake #4769: Drake build fails due to issues with string literals

rdeits, 2017-01-12:
Building drake master on ubuntu 14.04 is failing for me with:

```
[22/458] Building CXX object automotive/maliput/utility/test/CMakeFiles/generate_urdf_test.dir/generate_urdf_test.cc.o
FAILED: automotive/maliput/utility/test/CMakeFiles/generate_urdf_test.dir/generate_urdf_test.cc.o 
/usr/local/bin/c++   -DEIGEN_MPL2_ONLY -DGTEST_DONT_DEFINE_FAIL=1 -DGTEST_DONT_DEFINE_SUCCEED=1 -DGTEST_DONT_DEFINE_TEST=1 -DHAVE_SPDLOG -I/home/rdeits/locomotion/drake-distro/drake/.. -isystem /home/rdeits/locomotion/drake-distro/build/install/include -isystem /home/rdeits/locomotion/drake-distro/build/install/include/eigen3 -I/home/rdeits/locomotion/drake-distro/drake/thirdParty/bsd/spruce -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=extra -Werror=return-local-addr -Wno-unused-parameter -Wno-missing-field-initializers -O3 -DNDEBUG -fPIE   -std=gnu++14 -MMD -MT automotive/maliput/utility/test/CMakeFiles/generate_urdf_test.dir/generate_urdf_test.cc.o -MF automotive/maliput/utility/test/CMakeFiles/generate_urdf_test.dir/generate_urdf_test.cc.o.d -o automotive/maliput/utility/test/CMakeFiles/generate_urdf_test.dir/generate_urdf_test.cc.o -c /home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:89:3: error: missing terminating " character
   EXPECT_EQ(R"R(<?xml version="1.0" ?>
   ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:90:1: error: stray ‘\’ in program
 <robot name="dut">
 ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:90:1: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:91:3: error: stray ‘\’ in program
   <link name="world"/>
   ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:91:3: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:93:3: error: stray ‘\’ in program
   <joint name="world_to_road_joint" type="continuous">
   ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:93:3: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:94:5: error: stray ‘\’ in program
     <origin rpy="0 0 0" xyz="0 0 0"/>
     ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:94:5: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:95:5: error: stray ‘\’ in program
     <parent link="world"/>
     ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:95:5: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:96:5: error: stray ‘\’ in program
     <child link="surface"/>
     ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:96:5: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:99:3: error: stray ‘\’ in program
   <link name="surface">
   ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:99:3: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:101:5: error: stray ‘\’ in program
     <visual name="v1">
     ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:101:5: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:102:7: error: stray ‘\’ in program
       <origin rpy="0 0 0" xyz="0 0 0"/>
       ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:102:7: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:104:9: error: stray ‘\’ in program
         <mesh filename="junk.obj" scale="1.0 1.0 1.0"/>
         ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:104:9: error: missing terminating " character
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:109:1: error: stray ‘\’ in program
 )R", actual_urdf_contents);
 ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc: In member function ‘virtual void drake::maliput::utility::GenerateUrdfTest_AtLeastRunIt_Test::TestBody()’:
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:90:1: error: expected primary-expression before ‘<’ token
 <robot name="dut">
 ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:90:2: error: ‘robot’ was not declared in this scope
 <robot name="dut">
  ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:109:2: error: expected ‘)’ before ‘R’
 )R", actual_urdf_contents);
  ^
/home/rdeits/locomotion/drake-distro/drake/automotive/maliput/utility/test/generate_urdf_test.cc:89:29: error: expected ‘)’ before ‘;’ token
   EXPECT_EQ(R"R(<?xml version="1.0" ?>
```

I have:

```
g++ (Ubuntu 4.9.4-2ubuntu1~14.04.1) 4.9.4
cmake version 3.5.2
ninja version 1.7.2
```

rdeits, 2017-01-12:
It looks like the R-string literal isn't being handled correctly. Is this some feature that my compiler doesn't support?

liangfok, 2017-01-12:
I believe this may be a duplicate of this: https://drakedevelopers.slack.com/archives/build/p1484167906000558

liangfok, 2017-01-12:
Looks like that discussion recommends adding the following environment variable:

```
⁠⁠⁠⁠export CCACHE_CPP2=1﻿⁠⁠⁠⁠ 
```

rdeits, 2017-01-12:
Yup, agreed. Trying that now...

Thanks!

rdeits, 2017-01-12:
That worked, thanks!

closed 2017-01-12 by rdeits

liangfok, 2017-01-12:
This is probably another issue we should add to the FAQ: http://drake.mit.edu/faq.html.