RobotLocomotion/drake #4789: Bad UX When LeafSystem's Output Port Declarations and AllocateOutput() Do Not Match

liangfok, 2017-01-16:
# Problem Definition

When a user implements a defective `LeafSystem` whose set of output ports does not match what's allocated in `AllocateOutput()`, the following abort occurs while attempting to start the simulation:

```
abort: failure at /Users/liang/dev/drake-distro-4/drake/../drake/systems/framework/system_output.h:188 in get_mutable_port(): assertion 'index >= 0 && index < get_num_ports()' failed.
Process 22837 stopped
```

From this error message, it's difficult to determine that the root-cause is a faulty `AllocateOutput()` method.

# Suggested Course of Action

Can `System::AllocateOutput()` implement [NVI](https://en.wikibooks.org/wiki/More_C%2B%2B_Idioms/Non-Virtual_Interface) and check for correctness of a `System::DoAllocateOutput()`?

Or, is there a way to check for `AllocateOutput()`'s correctness while calling `DiagramBuilder::Build()` or `DiagramBuilder::BuildInto()`?

david-german-tri, 2017-01-17:
Agreed, we should improve this.

liangfok, 2017-01-17:
@david-german-tri, do you think the NVI approach mentioned in the OP is viable? Once I get your feedback, I'll reassign this issue to myself for resolution.

sherm1, 2017-01-17:
@liangfok, I'm reworking output port allocation to address #2890 and caching. If this isn't urgent you may want to wait until that lands.

liangfok, 2017-01-17:
@sherm1: Thanks for the note. I will stand by.

david-german-tri, 2017-04-18:
Reassigning to @sherm1, then.

jwnimmer-tri, 2017-05-24:
Does #6038 fix this?  Maybe that PR should gain the magic auto-close keyword stuff if so.

sherm1, 2017-05-24:
I think it does, but it would be great to have @liangfok verify that the UX is better before closing.

Also, does the auto-close magic work when added later? I know it does if included in the initial PR description but I *think* it is ignored in later comments.

jwnimmer-tri, 2017-05-24:
Right, you would have to edit the description, and then it works fine.

liangfok, 2017-05-24:
I will look into this ASAP.

liangfok, 2017-05-24:
#5348 made `LeafSystem::AllocateOutput()` final, but the problem persisted because leaf system code would still need to implement `LeafSystem::AllocateOutputVector()` and `LeafSystem::AllocateOutputAbstract()` meaning an output port can still not be properly allocated.

#6038 does away with `LeafSystem::AllocateOutputVector()` and `LeafSystem::AllocateOutputAbstract()` and implements the output port allocation in `LeafSystem::DoAllocate()` using either a user-supplied allocation function or a model-value object. Since all of the `LeafSystem::DeclareVectorOutputPort()` and `LeafSystem::DeclareAbstractOutputPort()` methods require the user to specify a model-value or allocator, it's no longer possible for an output port to not be allocated. Thus, I conclude that this issue will be resolved by #6038.

@sherm1, please feel free to add "Resolves #4789" to the original description of #6038.

liangfok, 2017-05-24:
Looking in `LeafOutputPort<T>::DoAllocate()`, it looks like it's still possible for a user to supply an allocation function that returns `nullptr`, which would result in the simulator to crash. I will test this hypothesis.

sherm1, 2017-05-24:
OK, thanks @liangfok - I'll wait.

liangfok, 2017-05-24:
@sherm1, I created the following unit test that shows how a user could cause the system to seg fault by providing a faulty output allocator:

https://github.com/liangfok/drake/commit/101e376e97568907fc88ed6eb20e60a87468454b

Are there any simple changes that could be applied to protect against this failure mode?

sherm1, 2017-05-25:
Thanks, @liangfok -- I'll see if I can get something more informative to come out!

sherm1, 2017-05-31:
@liangfok -- the current version of PR #6038 catches null allocators. When that lands you *should* be able to close this issue. It also attempts to test for consistency of non-null types, but that required some expensive stuff I could only afford in Debug builds. There are unit tests in output_port_test.cc.

liangfok, 2017-06-02:
@sherm1: Thanks! I verified that #6038 handles faulty allocator scenarios by aborting instead of seg faulting. Thus, please feel free to update #6038's description to indicate that it resolves this issue. 

BTW, the aforementioned abort currently happens when `Simulator` is instantiated, as opposed to when the faulty system is instantiated. If the system framework could abort when the faulty system is instantiated, that would be even better.

Here's my dev branch with the updated test case: https://github.com/liangfok/drake/tree/feature/separate-output_ports2-bad-allocator

sherm1, 2017-06-02:
Thanks, @liangfok! The bad allocator is noticed the first time it gets called, which occurs in CreateDefaultContext for a Diagram. In your example the Simulator is first to ask for a Context so that's when the first call to the allocator gets made. It's not clear to me how to catch it earlier -- I can't make a test invocation at the time the allocator is declared (in a System constructor) because the allocator does not have to be valid until the System has been completely built (which could mean reading in a urdf or whatever before sizing things).

liangfok, 2017-06-02:
OK, makes sense. Leaving the current behavior as-is is perfectly fine. Thanks again!

PR #6038 (2017-06-14): Separate output ports for independent allocation and calculation
OutputPortDescriptor gets an upgrade to a full object called OutputPort, which supports three functions:
- allocate
- calculate (unconditional)
- evaluate (cached)

Resolves #2890 (along the lines proposed in [this comment](https://github.com/RobotLocomotion/drake/issues/2890#issuecomment-268889174)).
Resolves #4789.

Specifying an OutputPort requires specifying an allocator function and calculator function with matching types. The allocator can be provided as an explicit function or by providing a model value that can be copied  [... cut]
closed 2017-06-14 by sherm1