RobotLocomotion/drake #4793: xcode build fails with fcl enabled

RussTedrake, 2017-01-16:
```
CompileC /Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/Objects-normal/x86_64/fcl_model.o multibody/collision/fcl_model.cc normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
    cd /Users/russt/drake-distro/drake
    export LANG=en_US.US-ASCII
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x c++ -arch x86_64 -fmessage-length=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit=0 -Wno-trigraphs -fpascal-strings -O0 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-non-virtual-dtor -Wno-overloaded-virtual -Wno-exit-time-destructors -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-shorten-64-to-32 -Wno-newline-eof -Wno-c++11-extensions -DCMAKE_INTDIR=\"Debug\" -DdrakeCollision_EXPORTS -DBULLET_COLLISION -DHAVE_SPDLOG -DEIGEN_MPL2_ONLY -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Winvalid-offsetof -mmacosx-version-min=10.11 -g -Wno-sign-conversion -Wno-infinite-recursion -Wno-move -I/Users/russt/drake-distro/build/drake/multibody/collision/Debug/include -I/Users/russt/drake-distro/drake/.. -I/Users/russt/drake-distro/build/install/include/bullet -I/Users/russt/drake-distro/build/install/include -I/Users/russt/drake-distro/build/install/include/eigen3 -I/Users/russt/drake-distro/drake/thirdParty/bsd/spruce -I/Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/DerivedSources/x86_64 -I/Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/DerivedSources -Wmost -Wno-four-char-constants -Wno-unknown-pragmas -F/Users/russt/drake-distro/build/drake/multibody/collision/Debug -Werror=all -Werror=ignored-qualifiers -Werror=overloaded-virtual -Werror=shadow -Werror=inconsistent-missing-override -Werror=sign-compare -Werror=return-stack-address -Wno-delete-non-virtual-dtor -Wno-shadow -fPIC -std=gnu++14 -MMD -MT dependencies -MF /Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/Objects-normal/x86_64/fcl_model.d --serialize-diagnostics /Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/Objects-normal/x86_64/fcl_model.dia -c /Users/russt/drake-distro/drake/multibody/collision/fcl_model.cc -o /Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/Objects-normal/x86_64/fcl_model.o
In file included from /Users/russt/drake-distro/drake/multibody/collision/fcl_model.cc:3:
In file included from /Users/russt/drake-distro/build/install/include/fcl/fcl.h:43:
In file included from /Users/russt/drake-distro/build/install/include/fcl/broadphase/broadphase_SSaP.h:42:
In file included from /Users/russt/drake-distro/build/install/include/fcl/broadphase/broadphase_collision_manager.h:44:
In file included from /Users/russt/drake-distro/build/install/include/fcl/narrowphase/collision_object.h:43:
In file included from /Users/russt/drake-distro/build/install/include/fcl/geometry/collision_geometry.h:45:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/motion/motion_base.h:41:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_matrix.h:41:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_vector.h:42:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model.h:143:
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:278:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0, c1, c2, c3;
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:278:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0, c1, c2, c3;
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:278:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0, c1, c2, c3;
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:278:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0, c1, c2, c3;
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:279:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0b = other.coeffs_[0], c1b = other.coeffs_[1], c2b = other.coeffs_[2], c3b = other.coeffs_[3];
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:279:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0b = other.coeffs_[0], c1b = other.coeffs_[1], c2b = other.coeffs_[2], c3b = other.coeffs_[3];
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:279:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0b = other.coeffs_[0], c1b = other.coeffs_[1], c2b = other.coeffs_[2], c3b = other.coeffs_[3];
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:279:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S c0b = other.coeffs_[0], c1b = other.coeffs_[1], c2b = other.coeffs_[2], c3b = other.coeffs_[3];
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:289:3: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
  register S tempVal = coeffs_[1] * c3b + coeffs_[2] * c2b + coeffs_[3] * c1b;
  ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:371:5: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
    register S a = -coeffs_[1] / (2 * coeffs_[2]);
    ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:376:7: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
      register S t = t0;
      ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:395:7: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
      register S t = t0;
      ^~~~~~~~~
/Users/russt/drake-distro/build/install/include/fcl/math/motion/taylor_model/taylor_model-inl.h:408:5: warning: 'register' storage class specifier is deprecated and incompatible with C++1z [-Wdeprecated-register]
    register S t = t0;
    ^~~~~~~~~
In file included from /Users/russt/drake-distro/drake/multibody/collision/fcl_model.cc:3:
In file included from /Users/russt/drake-distro/build/install/include/fcl/fcl.h:48:
In file included from /Users/russt/drake-distro/build/install/include/fcl/broadphase/broadphase_dynamic_AABB_tree.h:44:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/bv/utility.h:59:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/bv/utility-inl.h:45:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/bv/kIOS.h:41:
In file included from /Users/russt/drake-distro/build/install/include/fcl/math/bv/OBB.h:170:
[... cut ...]
 clang diagnostic ignored "-Wmaybe-uninitialized"
                          ^
17 warnings and 3 errors generated.

** BUILD FAILED **

The following build commands failed:
	CompileC /Users/russt/drake-distro/build/drake/multibody/collision/drake.build/Debug/drakeCollision.build/Objects-normal/x86_64/fcl_model.o multibody/collision/fcl_model.cc normal x86_64 c++ com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)
make: *** [/Users/russt/drake-distro/build/drake-prefix/src/drake-stamp/Debug/drake-build] Error 65

** BUILD FAILED **
```

jamiesnape, 2017-01-17:
Not sure what to do about Xcode builds. We could fix CMake so that Xcode builds can handle `-isystem`, but Xcode support for Bazel is limited (and I have never managed to get it to work) and very different to CMake.

jamiesnape, 2017-01-17:
(FCL is disabled for the Xcode CI builds)

sherm1, 2017-01-17:
@jamiesnape, do you think you could fix this upstream?

jamiesnape, 2017-01-17:
Yes, I have started to do so.

jamiesnape, 2017-01-19:
I will have a PR into FCL shortly that will fix this. Their master is currently broken, which I also need to fix. Both of those will then unblock getting the FCL collision model populated.

RussTedrake, 2017-03-14:
disabling xcode in favor of bazel.

closed 2017-03-14 by RussTedrake