RobotLocomotion/drake #4794: Mysterious Bazel CI Error: soft_failure.bzl: @drake_visualizer//:drake-visualizer does not work because drake-visualizer was missing.

liangfok, 2017-01-17:
# The Error

The following is a transient error. Re-running the CI may not reproduce the problem.

```
WARNING: _bazel_ubuntu/7d79f863c59b67ba7a1ab1641300f288/external/drake_visualizer/BUILD:2:48: soft_failure.bzl: @drake_visualizer//:drake-visualizer does not work because /.../build/install/bin/drake-visualizer was missing.
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/job/linux-xenial-gcc-bazel-experimental/536/

jwnimmer-tri, 2017-01-17:
I am not sure why that build was marked as failed.  With #4749 fixed, that warning message should not be upgraded to an error.  So either #4749 isn't really fixed, or something else went wrong with that build.

david-german-tri, 2017-04-13:
Jenkins has aged out the build logs, so I'm going to close this as wontfix.  Please reopen (and reassign to @stonier) if it recurs.

closed 2017-04-13 by david-german-tri