RobotLocomotion/drake #4796: Consider Amending the Style Guide to Require All Acronyms Be Treated As Single Words

liangfok, 2017-01-17:
The style guide currently has the following sentence for function names:

> Prefer to capitalize acronyms as single words (i.e. StartRpc(), not StartRPC()).

https://google.github.io/styleguide/cppguide.html#Function_Names

I've just encountered a constant, `kRPYValkyrieDoF`, in #4754 that does not abide by this rule. Should we amend the style guide to require constants to also abide by this rule, meaning this variable should be called `kRpyValkyrieDof`?

liangfok, 2017-01-17:
@david-german-tri I assigned this to you since you're today's on-call platform reviewer. I'll seek wider comment on Slack, however.

RussTedrake, 2017-01-17:
I think the current language in the style guide is sufficient to have recommended that change.  I don't think it warrants an explicit exception.  The more exceptions we have, the less likely people are to read and grok them.   I also don't find the use you cite to be problematic nor offensive.

liangfok, 2017-01-17:
OK, I wanted to confirm that it's OK to apply a sentence that appears in the "Function Names" section to constants. The type name section is consistent in the treatment of acronyms as single words:

https://google.github.io/styleguide/cppguide.html#Type_Names

sherm1, 2017-01-17:
I agree with @RussTedrake that (a) the styleguide already likes acronyms like `Url` rather than `URL` in all names, and (b) `RPY` or `Rpy` both seem inoffensive.

david-german-tri, 2017-01-17:
Since this is assigned to me, and I agree with @RussTedrake and @sherm1, I'm closing it.

closed 2017-01-17 by david-german-tri