RobotLocomotion/drake #4799: Consider Making Gravity Settable

liangfok, 2017-01-17:
# Problem Definition

Gravity currently is [hard-coded](https://github.com/RobotLocomotion/drake/blob/6cae3e4c08a0b078d52a6cf31d0fe6fd06b0d277/drake/multibody/rigid_body_tree.cc#L106) in `RigidBodyTree`'s constructor as follows:

```
a_grav << 0, 0, 0, 0, 0, -9.81;
```

This presumes we're controlling and/or simulating something where the acceleration due to gravity is (1) along the world frame's +Z axis and (2) has a value of -9.8 m/s^2, which might not be true if we're, for example, in space or on the moon. This issue tracks the discussion on whether gravity should be made a user-settable value.

This issue was created in response to @naveenoid's comment here: https://reviewable.io/reviews/robotlocomotion/drake/4682#-Kag3VzEtfVwW9Xi_Foj

sherm1, 2017-01-17:
We will be more flexible in MultibodyTree. I don't think we should fix this in RBTree unless there is an immediate need.

/cc @amcastro-tri

jwnimmer-tri, 2017-01-17:
This request seems of low value to curate in the issues database.

liangfok, 2017-01-17:
@amcastro-tri do you have a design doc for `MultiBodyTree`? If so, maybe these kinds of requests could be aggregated in such a document?

amcastro-tri, 2017-01-17:
No document. I'm working on a prototype that will be curated on a first iteration with the dynamics team before opening it up for discussion. 
Setting gravity (arbitrary direction, value) will just be a natural consequence of the MBT design. I actually see it as an advantage to test the "isotropy of space" (Landau loves that terminology) through unit testing.

liangfok, 2017-01-17:
Excellent. I trust you will eventually address this issue in MBT and will close this issue now.

closed 2017-01-17 by liangfok