RobotLocomotion/drake #4806: Build Problem on OS X - 'GL/glut.h' file not found

liangfok, 2017-01-18:
# The Error

I encountered the following error on OSX 10.11.6 using the latest on master (SHA e4c29605462fff36b5ba24ba1639f7299076ae65):

```
FAILED: src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkLCMGLProp.cxx.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DGLUT_AVAILABLE -DvtkDRCFilters_EXPORTS -I/usr/local/opt/vtk5/include/vtk-5.10 -I/Users/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters -Isrc/vtk/DRCFilters -I/Users/liang/dev/drake-distro-1/build/install/include/eigen3 -I/Users/liang/dev/drake-distro-1/build/install/include -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/include/python2.7 -g -fPIC   -F/usr/local/Cellar/qt/4.8.7_2/lib -MMD -MT src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkLCMGLProp.cxx.o -MF src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkLCMGLProp.cxx.o.d -o src/vtk/DRCFilters/CMakeFiles/vtkDRCFilters.dir/vtkLCMGLProp.cxx.o -c /Users/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/vtkLCMGLProp.cxx
/Users/liang/dev/drake-distro-1/externals/director/src/vtk/DRCFilters/vtkLCMGLProp.cxx:27:10: fatal error: 'GL/glut.h' file not found
#include <GL/glut.h>
         ^
```

This was after updating my system based on [these instruction](http://drake.mit.edu/homebrew.html) and doing a clean build.

liangfok, 2017-01-18:
I'm upgrading this issue to **emergency** because it's also showing up on CI.

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/

jamiesnape, 2017-01-18:
This is in Director itself: https://github.com/RobotLocomotion/director/commit/5de7e3e1e08f6cef278f36cf93608eeef1736b94

jamiesnape, 2017-01-18:
GLUT is at `<GLUT/glut.h>` on macOS / OS X, though it is deprecated.

PR #4811 (2017-01-18): Update director submodule SHA
Fixes #4806.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4811)
closed 2017-01-18 by jamiesnape