RobotLocomotion/drake #4807: Unable to Run CMake on Drake on Ubuntu 16.04 with VTK 6.2 and ROS Kinetic

liangfok, 2017-01-18:
# Problem Definition

The following error occurs when running `cmake ..` on an machine with Ubuntu 16.04, ROS Kinetic, and VTK6:

```
-- Preparing to build director with dependencies bot_core_lcmtypes;ctk_python_console;drake;lcm;libbot;pythonqt;qt_property_browser
CMake Warning at cmake/externals.cmake:208 (find_package):
  Could not find a configuration file for package "VTK" that is compatible
  with requested version "5.10".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/vtk-6.2/VTKConfig.cmake, version: 6.2.0

Call Stack (most recent call first):
  cmake/externals.cmake:522 (drake_add_cmake_external)
  CMakeLists.txt:312 (drake_add_external)

CMake Error at cmake/externals.cmake:210 (find_package):
  Could not find a configuration file for package "VTK" that is compatible
  with requested version "5.8".

  The following configuration files were considered but not accepted:

    /usr/lib/cmake/vtk-6.2/VTKConfig.cmake, version: 6.2.0

Call Stack (most recent call first):
  cmake/externals.cmake:522 (drake_add_cmake_external)
  CMakeLists.txt:312 (drake_add_external)

-- Configuring incomplete, errors occurred!
```

I suspect this started to happen after #4527 and the removal of the `-DUSE_SYSTEM_VTK=OFF` option.

patmarion, 2017-01-19:
You have to compile VTK5.10 from source, then set the VTK_DIR cmake variable in Drake's superbuild to the build or install location of VTK5.10.  The existing documentation in Drake regarding USE_SYSTEM_VTK is now out of date.

jwnimmer-tri, 2017-01-19:
/CC @david-german-tri FYI on the above; we could obey `VTK_DIR` for the Bazel VTK support as well.

mwoehlke-kitware, 2017-01-19:
Note: my plan is to make VTK available as an external, but default to using the system version. When this happens, I'll also update the documentation accordingly. Given #4834, I think we're going to need that anyway (getting it via director would no longer have worked anyway due to a circular dependency).

PR #4844 (2017-01-23): Add VTK as an optional external
Teach the superbuild that VTK is an optional system dependency, which defaults to using the system version, and how to build it if enabled but not using the system version.

This enables us to provide our own VTK if the system version is not available (fixes #4807). This is also much easier to use than the old method of relying on the director superbuild (which we no longer use) to build VTK, and unlike that approach will also allow drake itself (#4834) to use VTK.

Also, false coupling between director and signalscope is now fully broken.

---
This change is [[ [... cut]
closed 2017-01-23 by mwoehlke-kitware