RobotLocomotion/drake #4808: Mysterious VTK5 Error: ERROR: In vtkLookupTable.cxx: vtkLookupTable (0x2585670): Bad table range: [1e+299, -1e+299]

liangfok, 2017-01-18:
# The Error

```
03:30:57 4: ERROR: In /build/buildd/vtk-5.8.0/Common/vtkLookupTable.cxx, line 116
03:30:57 4: vtkLookupTable (0x2585670): Bad table range: [1e+299, -1e+299]
03:30:57 4: 
03:30:57 4: ERROR: In /build/buildd/vtk-5.8.0/Common/vtkLookupTable.cxx, line 116
03:30:57 4: vtkLookupTable (0x2584660): Bad table range: [1e+299, -1e+299]
03:30:57 4: 
03:30:58 4: ERROR: In /build/buildd/vtk-5.8.0/Filtering/vtkStreamingDemandDrivenPipeline.cxx, line 957
03:30:58 4: vtkStreamingDemandDrivenPipeline (0x233f820): The update extent specified in the information for output port 0 on algorithm vtkImageMapToColors(0x2305730) is 0 919 0 545 0 0, which is outside the whole extent 0 605 0 579 0 0.
```

# Examples

* https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-nightly-open-source-debug/86/

mwoehlke-kitware, 2017-01-18:
Why is this "emergency"? In my experience, VTK squawks, but the test passes anyway; is this not the case? (I don't see any failing CI builds?)

I would advise asking @patmarion to take a look, and drop the priority to medium, at most. (I might even go so far as to close the drake issue and re-file this against director...)

liangfok, 2017-01-18:
I labeled it emergency b/c I thought it was causing nightly CI to fail. Not sure if this is correct.

liangfok, 2017-01-18:
The three links in the OP are two nightly-CI tests that "fail" in the sense that they are yellow. I'm now looking to see if there was some other culprit for them being yellow.

patmarion, 2017-01-18:
That's the expected output for the test.  Here is Director's CI output for this test:

http://kobol.csail.mit.edu/cd/testDetails.php?test=3705673&build=26192

Some tests are supposed to print error messages 👍

liangfok, 2017-01-18:
I'm still looking through [this console log](https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-nightly-open-source-debug/86/) to identify why it turned yellow, but I found the following two things that might be worth mentioning:

```
03:14:47 CMake Warning at cmake/externals.cmake:208 (find_package):
03:14:47   Could not find a configuration file for package "VTK" that is compatible
03:14:47   with requested version "5.10".
03:14:47 
03:14:47   The following configuration files were considered but not accepted:
03:14:47 
03:14:47     /usr/lib/vtk-5.8/VTKConfig.cmake, version: 5.8.0
03:14:47 
03:14:47 Call Stack (most recent call first):
03:14:47   cmake/externals.cmake:522 (drake_add_cmake_external)
03:14:47   CMakeLists.txt:312 (drake_add_external)
```

and:

```
CTest Result: UNSTABLE DUE TO GIT FAILURES
```

and:

```
CMake Error: Some required settings in the configuration file were missing:
```

patmarion, 2017-01-18:
It's here in the full log: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/linux-clang-nightly-open-source-debug/86/consoleFull

```
03:18:36 fatal: The remote end hung up unexpectedly
03:18:36 Clone of 'https://github.com/RobotLocomotion/eigen-mirror.git' into submodule path 'externals/eigen' failed
```

patmarion, 2017-01-18:
btw, @mwoehlke-kitware is it possible to do the FindPackage for VTK 5.10 quietly to resolve the configuration warning?

liangfok, 2017-01-18:
Thanks! In this case, this issue is a duplicate of #4809.

closed 2017-01-18 by liangfok