RobotLocomotion/drake #4824: "Term" is ambiguous in Polynomial class

hongkai-dai, 2017-01-19:
In `Polynomial` class, we defined `Term` as "An individual variable raised to an integer power; e.g. x**2.". This is not a commonly accepted definition for `Term`. For example, in [Wikipedia](https://en.wikipedia.org/wiki/Polynomial#Definition), `Term` is defined as 
"Each term consists of the product of a number—called the coefficient of the term[2]—and a finite number of indeterminates, raised to nonnegative integer powers." i.e., it is the same as a monomial.

I propose to rename `Term` as `Exponentiation` or `VariableExponentiation` to avoid the ambiguity.

@RussTedrake what do you think?

RussTedrake, 2017-01-19:
As you've written it, I guess it's a univariate monomial?  (but perhaps you might not want to use that term since its intended use is probably as a component of a multivariate monomial?)

@frankpermenter -- any better suggestions on names?

hongkai-dai, 2017-01-19:
Exactly as you said, we use this univariate monomial as the building block for multivariate monomial. Besides, we define the monomial as `2 * x^2*y^3`, namely with a coefficient constant. This "Term" does not have a constant coefficient, it is in the form of `x^2`, so I do not quite want to call it univariate monomial, as that should mean `2 * x^2` according to our definition of monomial.

sherm1, 2017-01-30:
@hongkai-dai says this class is going to be history soon so I'm closing this.

closed 2017-01-30 by sherm1