RobotLocomotion/drake #4833: Can we delete compareParsers?

jwnimmer-tri, 2017-01-19:
Can we delete the `compareParsers` test at https://github.com/RobotLocomotion/drake/blob/master/drake/matlab/systems/plants/test/compareParsers.m?  It causes many headaches, the newest of which is wanting to use Boost in the new Multibody parser in #4831.

/CC @RussTedrake @sherm1 @SeanCurtis-TRI @edrumwri @amcastro-tri @liangfok.

SeanCurtis-TRI, 2017-01-19:
Can we enumerate any parsing functionality we know exists in matlab that does *not* yet exist in C++?  Although, in retrospect, that list is orthogonal to the test.  Clearly, whatever the test compares between the parsed results must not include differences, otherwise they wouldn't pass. :)

Generally, I'm find with seeing this go away.

liangfok, 2017-01-19:
The following comment affirms removal since the c++ parsers will be moving forward and thus diverge away from the MATLAB parsers: https://github.com/RobotLocomotion/drake/issues/2034#issuecomment-269628599

I support the removal of `compareParsers.m`.

amcastro-tri, 2017-01-20:
The test only compares C++ vs. Matlab implementations. I think with the new effort started in #4831 we will have the chance to write actual unit tests that verify the correctness of the loaded models from a given file. 
I am also comfortable removing this test if that allows us to move forward.

jwnimmer-tri, 2017-01-23:
Sounds like consensus.  If anyone wants to steal this, please feel free.  Otherwise I will get to it ASAP.

PR #4867 (2017-01-24): Remove compareParsers test
We'd like to use Boost in the C++ parser soon, so we can't have mexfiles use the parser anymore.  Fixes #4833.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4867)
closed 2017-01-24 by sherm1