RobotLocomotion/drake #4838: Need to Add a Check To Verify Quaternion Floating Joint Configuration Is Valid

liangfok, 2017-01-20:
# Problem Definition

In #4837, I mistakenly initialized the quaternion floating joint's configuration to be all zeros. This is incorrect since the "W" term of the quaternion floating joint must be 1, not 0, in its zero configuration. Such a bug should be programmatically detectable; `RigidBodyTree` should detect the error and squawk. This ticket tracks the addition of such a check.

sherm1, 2017-01-20:
I'd love to see good state validation in MBTree; not so sure about RBTree repair. @amcastro-tri ?

liangfok, 2017-01-20:
Yes, I agree. I'll update the OP to say MBTree instead of `RigidBodyTree`.

amcastro-tri, 2017-01-23:
State validation for MBT would definitely be a nice feature to have. In the near future however I think we can avoid this kind of problems in MBT by providing the user a clean API to set default states. The problem @liangfok had is because in the RBT world users have direct access to the state vector as a raw array of numbers. When there are things like quaternions in this soup of numbers it is very easy (and temping) to initialize this vector to zero (where `[0, 0, 0, 0]` does not make sense for a quaternion). Users should not be allowed to do so (or at least be highly restricted) but they should call MBT methods to initialize their states.

sherm1, 2017-01-23:
Since the Context allows direct state modification we may still have to deal with this. At simulation `Initialize()` we can watch for this and consider allowing an all-zero state (xc) at initialization to mean identity orientation for quaternions. We could just replace (0,0,0,0) with (1,0,0,0) during initialization processing where we may have to do other state modifications anyway to satisfy other constraints. The advantage of that is that a user can simply set the state to zero to get a reasonable initial configuration; otherwise they have to set it to zero then hunt down any tree joints that use quaternions and fix those.

We could also just report an error but I'm not sure that's better in practice.

closed 2017-02-01 by amcastro-tri