RobotLocomotion/drake #4840: Drake's version of `lcm-logger` fails with missing shared library

rdeits, 2017-01-20:
On OSX and Ubuntu 14.04, when attempting to run the version of `lcm-logger` that Drake installs, I get errors with missing `liblcm` shared libraries:

(Ubuntu 14.04):

```
 error while loading shared libraries: liblcm.so.1: cannot open shared object file: No such file or directory
```

(OSX):

```
dyld: Library not loaded: @rpath/liblcm.1.dylib
```

Both of these are on Drake master (from today).

rdeits, 2017-01-20:
If I build `lcm` master outside of Drake's build system, then I don't see this error, so I would suspect that this is a problem with the way Drake is configuring the LCM build.

jwnimmer-tri, 2017-01-20:
@mwoehlke-kitware Could you offer any advice?

rdeits, 2017-01-20:
We just checked on @manuelli 's computer, and his `lcm-logger` works. However, `ldd` shows that it's picking up his systemwide version of `liblcm` from `/usr/local/lib`, which I don't have.

mwoehlke-kitware, 2017-01-20:
Hmm... same here on macOS. On my Linux, though, it looks like the installed binary is still RPATH'd.

How is this *supposed* to work? Normally, if you run an installed binary, it either needs to be RPATH'd or its libraries need to be in the library search path. Which of those is expected to be true for drake?

jwnimmer-tri, 2017-01-20:
@mwoehlke-kitware Did the de-PODS perhaps drop the rpath?  It used to work -- bisect the regressing commit and debug from there?

jamiesnape, 2017-01-20:
RPATH is only part of PODs that has not been removed.

david-german-tri, 2017-01-20:
FYI I can reproduce this on Ubuntu 14.04 also.

david-german-tri, 2017-01-20:
> Normally, if you run an installed binary, it either needs to be RPATH'd or its libraries need to be in the library search path. Which of those is expected to be true for drake?

I believe the answer to this question is: RPATH.

david-german-tri, 2017-01-20:
As a workaround until the CMake gets fixed, you can use Bazel to build lcm-logger.  From `drake-distro`:

```
$ bazel build @lcm//:lcm-logger
$ ./bazel-bin/external/lcm/lcm-logger
```

patmarion, 2017-01-23:
The autoconf build of lcm set the rpath of the installed lcm-logger binary to the install bin dir.

The lcm-logger in the cmake build dir has rpath set, so you can run that, or set LD_LIBRARY_PATH.

`./build/externals/lcm/lcm-logger/lcm-logger`

patmarion, 2017-01-23:
originally reported: https://github.com/lcm-proj/lcm/issues/119

closed 2017-01-24 by david-german-tri

RoboEvangelist, 2017-04-04:
how exactly do you add, for instance, liblcm.so to the BUILD file on Bazel? I read the comments above, but it's still not clear to me

jwnimmer-tri, 2017-04-04:
Running these commands should run `lcm-logger` successfully:
```
$ bazel build @lcm//:lcm-logger
$ ./bazel-bin/external/lcm/lcm-logger
```
If that fails somehow, then please open a new issue showing what you tried and how it failed.

Of if there's some other problem not relating to `lcm-logger`, then perhaps open a new issue for that?

PR #4883 (2017-01-24): Set RPATH for "installed" binaries
Set variables to tell CMake to set RPATH's in installed binaries. Propagate these to external projects.

Fixes #4840.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4883)