RobotLocomotion/drake #4841: Valkyrie simulation has sliding feet (and then falls over)

rdeits, 2017-01-20:
Running the C++ valkyrie simulator with `valkyrie_simulation` and the controller with `valkyrie_pd_ff_controller` results in the feet slowly sliding together (while in contact with the ground), and then the robot falling over. Am I doing something wrong? Is this not the right way to run the simulator and controller?

jwnimmer-tri, 2017-01-30:
/CC @siyuanfeng-tri Any thoughts?

sherm1, 2017-01-30:
@SeanCurtis-TRI is working on an improved stiction model (already running but not PRed yet). That's a good thing but there is no certainty that it is the actual cause of the problem! Investigation ongoing ...

siyuanfeng-tri, 2017-01-30:
@rdeits depends on which sim you run? I'd guess you run the pid controlled version in examples/Val? That one is doing PD + some constant nominal torque, once q deviates (feet slide away) from whereever that feedforward is generated, it falls over. If you run the qp controlled version, it should be fine, ./examples/QPInverseDynamicsForHumanoids/system/valkyrie_balancing_controller_system

hongkai-dai, 2019-03-25:
From Siyuan's comment, the solution is to use the qp controller, not the pid controller. I will close this issue.

closed 2019-03-25 by hongkai-dai