RobotLocomotion/drake #4843: Halfway use of clang-format is unsustainable

jwnimmer-tri, 2017-01-20:
Right now Drake recommends using clang-format sometimes, but we don't enforce that all code in tree is clang-format idempotent.

This is bad, because:

(1) Now when anyone runs clang-format on a file in order to fix some cpplint in their PR, it reformats things not even touched by the PR, thus destroying the original author's artisinally-crafted whitespace.  The only argument against clang-format-all-the-things is that people disagree with its choices; that is logically irreconcilable with allowing people to clang-format whole files that they didn't author.

(2) Without tool assistance our team (and open-source contributers submitting PRs) are not experienced enough know how to format code readably, reliably.  If you want, I will go point out dozens of authors' mistakes in PRs, but I was hoping not to have to shame them.  The cpplint alone tests are not enough to identify all of the problems.  We need a tool that is sharper that cpplint at spotting errors.  Leaving it to code review time is expensive.

Formatting defects should be fixed (or at least called out) by a tool.  So maybe we should find a better tool than clang-format and cpplint.

We can look around for better tools, but absent finding any I will resume the clang-format-all-the-things march again.

liangfok, 2017-01-20:
> Now when anyone runs clang-format on a file in order to fix some cpplint in their PR, it reformats things not even touched by the PR, thus destroying the original author's artisinally-crafted whitespace.

This is a key reason why I have been running `clang-format` increasingly less often over time -- Drake's code is increasingly becoming not idempotent to `clang-format` meaning running `clang-format` is increasingly making changes that are not relevant to the current PR. It's a snowball effect since I'm sure my new code is not idempotent to `clang-format`.

david-german-tri, 2017-01-20:
@liangfok, `clang-format-diff` may be useful; it only changes modified lines.  Example usage to clang-format the previous commit:

```
git diff -U0 HEAD^ | clang-format-diff-3.9 -i -p1
```

liangfok, 2017-01-21:
@david-german-tri: Thanks! I will check it out.

jwnimmer-tri, 2018-10-24:
Last week I had an idea to start to move the forward.  Formatting and layout of the pydrake C++ bindings code is often challenging for developers.  The code layout is atypical versus regular imperative C++ code, so we often haggle about its whitespace and have to nitpick it to death.  It might be a good starting point for opt-in required clang-format idempotency, on a package-by-package basis.  In time, perhaps the whole tree works this way, but we have to start somewhere.  I tried to start in `//common` last time and got shot down.  Possibly being more rigid about boilerplate code would fly better, and prove out the tooling.

SeanCurtis-TRI, 2023-08-18:
It's worth noting that this is blocked by #20011.

SeanCurtis-TRI, 2025-03-31:
We have various cases where our *linter* doesn't agree with our *formatter*.

1. `static_cast<void()>` passes the linter. But the formatter inserts a space: `static_cast<void ()>` that makes the linter angry again.
2. Something like this:
```c++
const Foo foo {
  .param_name = {
    {.a = 10, .b = 20},
    {.a = 30, .b = 40}
  },
};
```
passes the linter, but gets formatted to:
```c++
const Foo foo {
  .param_name =
    {
      {.a = 10, .b = 20},
      {.a = 30, .b = 40}
    },
};
```
which, again, makes the linter angry (about the `{` all by itself on a line).
3. For non-exported namespaces of the form: `namespace internal DRAKE_NO_EXPORT {`, the closing brace would be `}  //namespace internal` and make the linter happy. But the formatter will append the `DRAKE_NO_EXPORT` which makes the linter angry.

Note: I will keep editing this as I classify more such animals in the wild.

jwnimmer-tri, 2025-03-31:
The overall approach is to disable cpplint formatting diagnostics (especially whitespace), since clang-format is the one true god.  Ideally, our build system would do that automatically (using a different `CPPLINT.cfg` depending on whether clang-format-lint is enabled).  In the meantime, we can either put `// NOLINT` comments, or straight up edit `CPPLINT.cfg` to disable the warnings globally.

jwnimmer-tri, 2025-03-31:
When formatting problems crop up, let's just discuss them in the PRs instead of here.  That will provide better context for solving them.  The namespace problem, for example, was just a missing config setting in the clang-format config file.

jwnimmer-tri, 2025-08-26:
I've opened pull requests to remove the opt-out across the board.  After all of those land, I'll open a PR that removes the opt-out flag from our `lint.bzl` entirely.

closed 2025-08-28 by rpoyner-tri