RobotLocomotion/drake #4848: Split symbolic expression test

soonho-tri, 2017-01-21:
Suggested from https://github.com/RobotLocomotion/drake/issues/4755.

hongkai-dai, 2017-01-21:
@soonho-tri I was actually working on this for the last two days, to refactor `mathematical_program_test`. Currently it contains the tests for adding variables/constraints, and also testing different solvers, with or without the symbolic form.

I am refactoring this file to different files, that `mathematical_program_test` only contains the tests on adding variables/constraints (namely it does not rely on any solvers). Now `mathematical_program_test` tests adding constraints/costs with or without the symbolic form.

I am adding separate files testing each solver. When testing solvers, the problem is constructed with or without the symbolic form.

I should be able to open the PR tomorrow, and I will ask you to review.

hongkai-dai, 2017-01-21:
Ah I see, we are talking about different files. You are splitting the symbolic tests in `symbolic_expression_test.cc`, I am splitting the symbolic tests in `mathematical_program_test.cc`

soonho-tri, 2017-01-21:
:-) great to know that we don't work on the same thing! Mine is almost done as well. I'll make a PR soon.

> I should be able to open the PR tomorrow, and I will ask you to review.

Great!

closed 2017-01-23 by soonho-tri