RobotLocomotion/drake #4849: Use symbolic::Variable as a variable type in drake::Polynomial class

soonho-tri, 2017-01-21:
Suggested from https://github.com/RobotLocomotion/drake/issues/4755.

soonho-tri, 2017-01-24:
Currently, `Polynomial::VarType`  is `unsigned int`. I'd like to know if we can use `symbolic::Variable` for `Polynomial::VarType` instead.

Issues of the current design
----------------------------

First of all, it starts with the following work-flow in the `MathematicalProgram` framework where we can get a `Polynomial` from a `symbolic::Expression`. 
```

DecisionVariableScalar (which is symbolic::Variable)
       ||
       ||  Construct expression
       \/
symbolic::Expression 
       ||
       ||  Expression::ToPolynomial()
       \/
Polynomial
```

Note that when `Expression::ToPolynomial()` translates a `symbolic::Variable{Name, ID}` to a `Polynomial` it uses the **same** `ID` but drops `Name` part.

Because of this new way of constructing `Polynomial`, I've found the following potential issues:

1. There can be a name collision between the `Polynomial`s generated by `Expression::ToPolynomial()` and the `Polynomial`s constructed by using its constructors. For example:

    ```c++
    const Expression e{symvar_1};
    const Polynomiald p1{e.ToPolynomial()};
    const Polynomiald p2{"<var_name>"};  // not clear if p1 == p2 or not.
    const Polynomiald p3{p1 + p2};  // It can be 2 * p2, or just p1 + p2 
    ```
2. It's hard to re-construct `DecisionVariableScalar`s from a `Polynomial` generated by `Expression::ToPolynomial()` method. It's because we dropped the name part in the translation. There are `IdToVariableName` and `VariableNameToId` methods in `Polynomial` class but according to the documentation, `VariableNameToId` method only uses the first four characters in conversion, which sounds like a limitation.

soonho-tri, 2017-01-24:
@RussTedrake, @ggould-tri, @sammy-tri : It'd be great if you read the above comment before today's meeting. I'm sorry for the last-minute writing.

/cc @jwnimmer-tri , in case you are interested in this thread.

ggould-tri, 2017-01-24:
The variable interning approach in Polynomial is awful and should go away; I believe the only real requirements on VarType is that it can key sets and maps and can be tested for equality, so anything reasonably small should be able to act as VarType without difficulty just be changing the typedef.

My only question is whether it would make more sense to eliminate the Polynomial class entirely and instead just use Symbolic (or reimplement Polynomial as a wrapper around a restricted subset of Symbolic).

jwnimmer-tri, 2017-01-24:
> It's hard to re-construct DecisionVariableScalars from a Polynomial generated by Expression::ToPolynomial...

A lookup method from ID to DVS would be trivial, in case that helps.

soonho-tri, 2017-01-24:
@jwnimmer-tri , one of the decisions came out from the meeting is to replace the current Polynomial class with one based on `symbolic::Expression`. I'll work on that.

soonho-tri, 2017-01-24:
Close this issue because of https://github.com/RobotLocomotion/drake/issues/4878.

closed 2017-01-24 by soonho-tri