RobotLocomotion/drake #4862: Modify the KUKA iiwa simulation to publish joint torque, position, and velocity in a single LCM message

liangfok, 2017-01-23:
Currently, the KUKA iiwa simulation does not output torques due to the direct-feedthrough issue. This issue tracks the implementation of a workaround, which would enable a single message to contain all three values (joint position, velocity, and torque). It will likely include modifying the PID Controller to publish its commanded torque values over a local loop-back LCM channel.

sammy-tri, 2017-01-23:
Updated title to reflect the fact that the iiwa hardware does not publish velocities.

sammy-tri, 2017-01-23:
The naive (to me) way to fix this issue would be to make the PidControlledSystem which is generating the input torques to RigidBodyPlant have an extra output port for the internally calculated commanded torques.  This doesn't work because it creates a loop in the diagram (SimulatedKuka in this case) which contains the PidControlledSystem (because the gravity compensator is connected in a feedback loop) where PidControlledSystem::DoCalcOutput() tries to generate the values for both ports and the new one is direct feedthrough (even though it doesn't feed into the gravity compensator and need not be calculated).

One workaround might be to (optionally) cache the torque values in PidControlledSystem as discrete state with a period equal to the LCM transmission interval for the iiwa status message.  I'd need to think about this a bit more.

naveenoid, 2017-01-23:
Why not compute the torques in some kind of JointTorqueSensor System coupled with the RigidBodyPlant? This kind of a system would simply use its own RigidBodyTree (like GravityCompensator) and compute the inverseDynamics method. This would be the "correct" way to generate torque signals. I could implement such a system.

manuelli, 2017-01-24:
Why would you need to compute the torques again? Aren't they being fed into the rigid body plant at one of it's input ports?

liangfok, 2017-01-24:
_(The following was transcribed from a Slack conversation)_

* Having each torque sensor instantiate its own `RigidBodyTree` is too expensive and slow.
* There are cheaper ways to model non-oracular sensors like by adding noise and bias.
* What we really want is to modify `PidControlledSystem` to output `plant_input_adder`'s output, which contains the torque commands. See [this line](https://github.com/RobotLocomotion/drake/blob/a2b9c8c8100894806a43d7fbb3c78e47994f0cd8/drake/systems/controllers/pid_controlled_system.cc#L133) where the output of `plant_input_adder` is currently fed into the `RigidBodyPlant`.

sammy-tri, 2018-05-11:
I think this has been done since #7822 merged.

closed 2018-05-11 by sammy-tri