RobotLocomotion/drake #4873: Check if the variables in the symbolic expression are variables in the MathematicalProgram

hongkai-dai, 2017-01-23:
Currently when we add a symbolic expression as a constraint/cost into `MathematicalProgram`, we do not check if the variables in that symbolic expression are decision variables in the `MathematicalProgram`. 

BTW, what is our strategy when doing unit test on the failure cases? I can add a unit test that adding a symbolic expression with some variables NOT included in the mathematical program, and expect there is an error being thrown. I vaguely remember we do not check the failure case in the unit test, is that still true?

@soonho-tri  might be interested.

soonho-tri, 2017-01-23:
> I can add a unit test that adding a symbolic expression with some variables NOT included in the mathematical program, and expect there is an error being thrown.

I agree that this is the way to go.

>  I vaguely remember we do not check the failure case in the unit test, is that still true?

I've used `EXPECT_THROW` in the symbolic unit tests. Is there our policy about it?

hongkai-dai, 2017-01-23:
Great, I will try `EXPECT_THROW` approach. Should I identify the type of error being thrown?

soonho-tri, 2017-01-23:
> Should I identify the type of error being thrown?

Yes. Most of the time, I do that. For example,

```c++
EXPECT_THROW(<expression throwing std::runtime_error>, std::runtime_error);
```

soonho-tri, 2017-01-23:
Here is GTEST documentation for that : https://github.com/google/googletest/blob/master/googletest/docs/AdvancedGuide.md#exception-assertions

closed 2017-01-28 by hongkai-dai