RobotLocomotion/drake #4875: Drake Unable to Load SDF With Explicit Pin-To-World Joint

liangfok, 2017-01-24:
# Problem Definition

A model in an SDF can have a special joint that effectively pins the model to the world. An example of this is located here:

https://github.com/liangfok/drake/tree/feature/sdf_pinned_to_world/drake/multibody/models/box_pinned_to_world

When this model is loaded into Gazebo, it does not fall due to the effects of gravity since it is pinned to the world.

The problem is when this model is loaded into Drake, Drake aborts with the following exception:

> 160: C++ exception with description "No root bodies found. Every body referenced by the supplied list of body indices has a joint connecting it to some other body.  This will result in RigidBodyTree::compile() looping indefinitely. This scenario currently not supported in Drake." thrown in the test body.

Note that #1821 does not cover this scenario, which has a joint whose inboard body is the world, but no world body; it only covers the scenario when there is an explicit "world" link.

jwnimmer-tri, 2017-02-16:
@clalancette How much of an impediment is this to your work?  If there is a simple just-as-good fix to the models, great.  But if having a code fix here would be a meaningful improvement, I will prioritize this fix.

clalancette, 2017-02-16:
I don't think this should affect me too much, at least in the short term.  So don't bother prioritizing this for now.

amcastro-tri, 2018-06-11:
This actually is the main mechanism by which you fix bodies to the world in MBP. If no fixed joint connects your robot to the world, it'll start falling due to gravity, as in Gazebo.

closed 2018-06-11 by amcastro-tri