RobotLocomotion/drake #4882: Create monomial and monomial basis as symbolic expressions

hongkai-dai, 2017-01-24:
The API can look like
```C++
// A monomial is of the form x^2*y^3, it does not have the constant factor.
// To generate a monomial x^2*y^3, map_var_to_exponent contains the pair
// (x.get_id(), 2) and (y.get_id(), 3)
Expression Monomial(const unordered_map<size_t, int>& map_var_to_exponent);

// Return all monomials up to a given degree. The degree of a monomial is the summation of all
// the exponents for each variable. For example, x^2 * y ^3 has degree 5.
Eigen::Matrix<Expression, Eigen::Dynamic, 1> MonomialBasis(const Variables& vars, int degree);
```

hongkai-dai, 2017-01-24:
As @soonho-tri and I discussed, when calling `MonomialBasis` function, the order of the expressions inside the returned vector, depends on the order of the variables. For the first round of implementation, we will use the order defined by `Variable`. We can implement other orders later.

soonho-tri, 2017-01-24:
Just to check, do you expect `[x^2, xy, y^2, x, y]` for the return value of `MonomialBasis({x, y}, 2)`?

hongkai-dai, 2017-01-24:
Almost, I expect [x^2, xy, y^2, x, y, 1] as the return value of `MonomialBasis({x, y}, 2)`.

soonho-tri, 2017-01-24:
OK. Thanks for the check.

closed 2017-01-26 by soonho-tri