RobotLocomotion/drake #4884: make drake::symbolic::Variable just drake::Variable

soonho-tri, 2017-01-24:
The goal is to have a single notion of "variable" in drake code base, to avoid code duplication and possible name collisions.

soonho-tri, 2017-01-24:
/cc @hongkai-dai

soonho-tri, 2017-01-27:
@jwnimmer-tri , I'm working on moving `Variable`, `Variables`, and `Environment` classes from `common::symbolic` namespace to `common` namespace. Which one of the followings is the best way to shape them in `BUILD` file?

 1. Keep them under `symbolic` library
 2. Have each in its own library (i.e. `variable.cc` => `variable`)
 3. Have the three in one library (i.e. `variable.cc`, `variables.cc`, `environment.cc` => `variable`)

jwnimmer-tri, 2017-01-27:
For Bazel, the right answer is the smallest libraries that make sense as standalone components.

closed 2017-01-31 by soonho-tri