RobotLocomotion/drake #4885: Add type to symbolic::Variable

soonho-tri, 2017-01-24:
For now, all the `symbolic::Variable`s implicitly have the type `continuous`. We will have explicit `type_` member in a `symbolic::Variable`, namely `continuous`, `int`, `binary`.

soonho-tri, 2017-01-24:
/cc @hongkai-dai

hongkai-dai, 2017-01-24:
Currently in `MathematicalProgram`, we defined
```C++
enum class VarType { CONTINUOUS, INTEGER, BINARY };
```
I will clean that up after this issue is fixed.

jwnimmer-tri, 2017-01-24:
Is this characteristic really intrinsic to a symbolic variable?  Non-real variables are not a valid ScalarType for systems, so we'd have to be careful if we decide to admit more types for the purposes of the mathematic program API, that we don't mess up its use for the systems framework.

hongkai-dai, 2017-01-24:
The primary consumer for `symbolic::Variable` is mathematical optimization and formal verification. In both two cases there is the notion of variable type. Do we have other usage for `symbolic::Variable`?

soonho-tri, 2017-01-24:
> Non-real variables are not a valid ScalarType for systems

@jwnimmer-tri , How difficult is it to lift the restriction?

> The primary consumer for symbolic::Variable is mathematical optimization and formal verification. In both two cases there is the notion of variable type. Do we have other usage for symbolic::Variable?

@hongkai-dai , I believe that we want to instantiate systems with `<symbolic::Expression>` and , for example, extract internal dynamics of a system as `symbolic::Expression` for analysis.

soonho-tri, 2017-01-24:
@hongkai-dai , another use case of `symbolic::Expression` is @jadecastro 's Hybrid system design.

jwnimmer-tri, 2017-01-24:
All of the `drake/systems/framework` and various specific leaf systems are templated on `T`, and for many of them, the `T` is expected to admit `symbolic::Variable`.  One requirement on `T` is that it has real-valued semantics.  If we feed a `symbolic::Variable` tagged as `int` or `binary` into an `System::EvalOutput`, it's not clear how the code will realize that something is wrong, or even compile in the first place.

jwnimmer-tri, 2017-01-24:
>> Non-real variables are not a valid ScalarType for systems
>
> @jwnimmer-tri , How difficult is it to lift the restriction?

I wouldn't even know what it means for a non-real scalar to be used as an e.g. continuous output or state.  Of course, _Abstract_ state can be modal or integer-valued or boolean, but `T` on its own is assumed to be real-valued.

hongkai-dai, 2017-01-24:
Will it help if we have three classes `RealVariable`, `IntegerVariable` and `BinaryVariable`? If this feels too much of a hassle, then I think we can just keep whatever we have now, without adding type information to `Variable`.

soonho-tri, 2017-01-24:
It seems that we just need to keep the current solution where we keep `Variable` type-free and a user of this class keeps a mapping from a Variable to its type internally.

jwnimmer-tri, 2017-01-24:
To be clear `T` in the case of Systems is `= symbolic::Expression`, not just `Variable`.  We could possibly admit a tag on the Variables that is always "REAL" when used on a System; we just need to think about how to make sure Systems code is correct, fail-fast, and easily understood.

closed 2017-01-24 by soonho-tri